IanHarvey/bluepy #26: How to force pairing with a device with Bluepy?

louiscaron, 2015-01-15:
Hello,
I have successfully established an LE connection to my BLE device with Bluepy.  Now for testing reasons, I would like to control the authentication and encryption of the link.  Is this possible?  I have tried setting the security level but it did not change anything.  Is is possible to do this?  If this is the case, who controls the storage of the link keys?
Thanks

clarkli86, 2015-02-12:
It is possible to force pairing/bonding with the btle.setSecurityLevel() function. When I set the security level to medium after the connection is established, I could see the pairing request/response/key generation in hcidump and my bluetooth device.
<pre>
btle.connect()
btle.setSecurityLevel()
</pre>
equals
<pre>
gatttool> connect <mac>
gatttool> sec-level medium
</pre>

However, it is not clear to me who controls the storage of the link keys. (I assume it is the bluetoothd daemon)

DigitalGabriele, 2017-03-09:
Hej, 

I have been trying to set security level to medium through gatttool, but it does not react. Same goes if I try to use setSecurityLevel(). What could be the problem?

mdehoogh, 2017-10-26:
I didn't know that I could use setSecurityLevel() to that purpose so I adapted btle.py a bit so that I could call the pair method of bluepy-helper. First I got a nomgmt error, and seeing that mgmt_master would only be created when iface is not None I changed its default in Peripheral to 0 instead. Then, it was still not pairing presenting me with a protoerr result. So, any reason why the pairing won't work this way? (I did execute 'pairable on' beforehand.)

abhishek-v-pandey, 2021-06-16:
Hello @clarkli86 I have tried your method and used setSecurityLevel() method but it does not work. Even tried with gatttool command line but no help over there as well.

Can you please explain how you got connected ?