IanHarvey/bluepy #27: random address not working

spedemon, 2015-01-20:
I am using bluepy on Debian (on the Intel Edison i386 board). While there is no problem connecting to BLE devices that expose a 'public' address (it works just fine with the TI SensorTag and with the RedBearLab BLE Mini), there is a problem (__main__.BTLEException: Failed to connect to peripheral) connecting to devices that expose a 'random' address (e.g. the RedBearLab BLE Blend Micro). 
Please note that 'gatttool -b xx:xx:xx:xx:xx:xx -I' connects to a device with 'public' address and 'gatttool -b xx:xx:xx:xx:xx:xx -I -t random' to a device with 'random' address.

louiscaron, 2015-01-20:
Hello Spedemon,
I initially had the same problem.  The reason was that the btle.py does not expose the possibility to configure the address type of the device you want to connect to.  The good surprise is that the bluepy-helper tool does support it so it was very easy to add it.  Here is what I have changed in btle.py:
    def connect(self, addr, addrtype="public"):
        if len(addr.split(":")) != 6:
            raise ValueError("Expected MAC address, got %s", repr(addr))
        self._startHelper()
        self.deviceAddr = addr
        self.deviceAddrType = addrtype
        self._writeCmd("conn %s %s\n" % (addr, addrtype))
        rsp = self._getResp('stat')
        while rsp['state'][0] == 'tryconn':
            rsp = self._getResp('stat')
        if rsp['state'][0] != 'conn':
            self._stopHelper()
            raise BTLEException(BTLEException.DISCONNECTED,
                                "Failed to connect to peripheral %s" % addr)

And when I invoked the Peripheral constructor:
conn = Peripheral(devaddr, deviceAddrType = "random")

and that worked fine.  I have also added a few goodies to btle and to the bluepy helper to support disconnection and reconnection to the same device, this is useful to perform test automation.  We I get a chance, I will try to post those changes to the development repo.

Hvae you looked at the Intel Community forum for Edison before posting your question (https://communities.intel.com/community/makers/edison)?  If not, it may be worthwile for other Edison users to add you experience with bluepy in here (eventually a link to this forum thread).

Thanks
Louis

spedemon, 2015-01-20:
Hi Luis, 
Thank you for your help! I will test it tonight at post back here. How do you manage the reconnection? Perhaps it would be useful to start an issue regarding reconnection. 
Best, 
Stefano

spedemon, 2015-01-21:
Hi Luis, 

Thank you, that solves the problem, now this Python command is successful: 

conn      = btle.Peripheral("CA:5B:3F:37:1D:A1","random")

However, note that the following command causes an exception: 

python btle.py CA:5B:3F:37:1D:A1 random 

The error occurs when one requests the characteristics of the service with index 0: 

conn = btle.Peripheral("CA:5B:3F:37:1D:A1","random")
services =  conn.getServices() 
services[0].getCharacteristics() 

Is there a reason why this occurs with random addresses only? 

Stefano

louiscaron, 2015-01-21:
Ciao Stefano,
The 'random' notion is only used at connection time so if your connection was successful, you should correctly retrieve the services and characteristics.  
It is difficult for me with so little information to really help you.  A few hints:
- enable the debug logs in btle.py
- use sudo hcidump to catch all the hci commands to your BT controller, this could help you find the source of the problem
Good luck,
Louis
Quando avvrai trovato la soluzione, puoi condividerla sul forum Edison?  Grazie

louiscaron, 2015-01-22:
If you want to share your btle.py with me, I can help you fix it.

Louis

spedemon, 2015-01-22:
Hi Luis, thank you very much for your help. With the fix that you suggested, I can do what I need, however something is not right. After modifying btle.py according to your suggestion, the following runs  successfully with 'do_get_characteristics_service_0 = False' but fails with 'do_get_characteristics_service_0=True': 

import btle 
do_get_characteristics_service_0 = False 
mac = "CA:5B:3F:37:1D:A1"
conn = btle.Peripheral(mac,"random")
services = conn.getServices() 
for i in range(len(services)): 
    if i==0: 
        if do_get_characteristics_service_0: 
             ch =  services[i].getCharacteristics()
    else: 
        ch = services[i].getCharacteristics()

Grazie! Nei prossimi giorni cercherò di postare nel forum Edison. 
Stefano

louiscaron, 2015-01-22:
Hello Stefano,Since I do not have your setup, it is difficult for me to have an idea of the problem.  But I will try to eventually give you some hints:- what is the size of the "services" returned by conn.getServices(), is it 1 or something else- could there be a permission problem on the service 0 that would require pairing
I suggest again that you capture the full hci logs and study them.  I can do it for you if you want if you send them to me.
Louis

PrzemoF, 2015-01-30:
I had the same problem with heart rate sensor Tacx (device name: Tacx HRB 04741)
After changing "public" to "random":
`
pi@occberry ~/bluepy/bluepy $ git diff bluepy-helper.c
diff --git a/bluepy/bluepy-helper.c b/bluepy/bluepy-helper.c
index 26a3dd7..5b102e8 100644
--- a/bluepy/bluepy-helper.c
+++ b/bluepy/bluepy-helper.c
@@ -461,7 +461,7 @@ static void cmd_connect(int argcp, char **argvp)
                if (argcp > 2)
                        opt_dst_type = g_strdup(argvp[2]);
                else
-                       opt_dst_type = g_strdup("public");
+                       opt_dst_type = g_strdup("random");
        }
 
        if (opt_dst == NULL) {
@@ -963,7 +963,7 @@ int main(int argc, char *argv[])
 
        opt_src = NULL;
        opt_dst = NULL;
-       opt_dst_type = g_strdup("public");
+       opt_dst_type = g_strdup("random");
 
         printf("# " __FILE__ " built at " __TIME__ " on " __DATE__ "\n");
         fflush(stdout);
`
it seems to work fine:
`pi@occberry ~/bluepy/bluepy $ sudo python ./btle.py D6:90:A8:08:F0:E4
Connecting to: D6:90:A8:08:F0:E4
Service <uuid=Generic Attribute handleStart=8 handleEnd=11> :
    Characteristic <2a05>
    -> Error from Bluetooth stack (comerr)
Service <uuid=Generic Access handleStart=1 handleEnd=7> :
    Characteristic <Device Name>
    -> 'Tacx HRB 04741'
    Characteristic <2a01>
    -> 'A\x03'
    Characteristic <2a04>
    -> '\x90\x01 \x03\x00\x00\x90\x01'
Service <uuid=Battery Service handleStart=18 handleEnd=21> :
    Characteristic <Battery Level>
    -> 'd'
Service <uuid=Heart Rate handleStart=12 handleEnd=17> :
    Characteristic <2a37>
    -> Error from Bluetooth stack (comerr)
    Characteristic <2a38>
    -> '\x01'
Service <uuid=Device Information handleStart=22 handleEnd=65535> :
    Characteristic <Manufacturer Name String>
    -> 'Tacx'
    Characteristic <Model Number String>
    -> 'T1994'
    Characteristic <Hardware Revision String>
    -> 'V1.01'
    Characteristic <Firmware Revision String>
    -> 'V1.05'
    Characteristic <Software Revision String>
    -> 'V1.02'
`
I'm not sure what all those `Error from Bluetooth stack (comerr)` mean, but without the changes it was always ending up with time out error

IanHarvey, 2015-02-24:
@spedemon - I've merged @PrzemoF 's changes in commit ad2de26.
Could you confirm if this solves the original problem? If so I'll close the issue.
Thanks,
Ian

spedemon, 2015-03-03:
Hi Ian, 
Thanks a lot. I will try on Saturday and will write back! 
Stefano

closed 2015-09-30 by IanHarvey

hardwarewinjit, 2016-08-24:
Hi,
I am facing issue while using btle.py
Please find below error message while running code on Linux

   p = Peripheral("C8:0F:10:77:3C:38","public")
  File "/usr/lib/python2.7/site-packages/bluepy/btle.py", line 318, in __init__
    self.connect(deviceAddr, addrType, iface)
  File "/usr/lib/python2.7/site-packages/bluepy/btle.py", line 367, in connect
    "Failed to connect to peripheral %s, addr type: %s" % (addr, addrType))
bluepy.btle.BTLEException: Failed to connect to peripheral C8:0F:10:77:3C:38, addr type: public

Kindly help
Thank you,
Shruti