IanHarvey/bluepy #36: Support for Sensortag 2

vielmetti, 2015-03-23:
TI has a new Sensortag 2, about the same shape and size as the original Sensortag but with a couple more sensors, a new processor, and better battery life.

When I try to connect to it with sensortag.py I get this:

```
emv@rubus ~/bluepy/bluepy $ python ./sensortag.py --all 68:C9:0B:06:C9:08
Connecting to 68:C9:0B:06:C9:08
Traceback (most recent call last):
  File "./sensortag.py", line 255, in <module>
    tag.barometer.enable()
  File "./sensortag.py", line 128, in enable
    self.calChr = self.service.getCharacteristics(self.calUUID) [0]
IndexError: list index out of range
```

There's been some porting work to support the Sensortag 2 from node.js, and there's UUIDs and sample code etc pointed to from here: 

https://github.com/sandeepmistry/node-sensortag/issues/34

which should be a useful stop to sort through what has changed.

vielmetti, 2015-04-21:
As a followup:

node-sensortag now has a full set of Sensortag 2 support, which means I'm super-tempted to port that work to bluepy.

vielmetti, 2015-09-07:
Just an update - if I pick out some of the sensors on a Sensortag 2 they do work, e.g. humidity:

```
% python ./sensortag.py -H -t 60 68:C9:0B:06:C9:08
Connecting to 68:C9:0B:06:C9:08
('Humidity: ', (23.957094726562495, 66.2503662109375))
('Humidity: ', (24.01072021484375, 66.6165771484375))
```

IanHarvey, 2016-01-27:
This is now done - sensortag.py has CC2650 support in it (it will auto-detect between the 2650 and the original V1 tag). 

Cheers
Ian

closed 2016-01-27 by IanHarvey

vielmetti, 2016-01-28:
Thanks @IanHarvey !

booherbg, 2016-03-07:
Hi everyone -- quick question. I can't find reference to the voltage sensor in the bluepy project for the new sensortag. Is that supported?