IanHarvey/bluepy #37: Support for latest bluez source code tree

vielmetti, 2015-03-23:
I tried to build bluepy-helper against the current bluez source tree (bluez-5.29). It was straightforward to point the makefile at that source tree, but I got a raft of errors and did not pursue it further.

```
gcc -L. -O0 -g -DHAVE_CONFIG_H -I../bluez-5.29/attrib -I../bluez-5.29 -I../bluez-5.29/lib -I../bluez-5.29/src -I../bluez-5.29/gdbus -I../bluez-5.29/btio `pkg-config glib-2.0 dbus-1 --cflags` -o bluepy-helper bluepy-helper.c ../bluez-5.29/lib/bluetooth.c ../bluez-5.29/lib/hci.c ../bluez-5.29/lib/sdp.c ../bluez-5.29/lib/uuid.c ../bluez-5.29/attrib/att.c ../bluez-5.29/attrib/gatt.c ../bluez-5.29/attrib/gattrib.c ../bluez-5.29/attrib/utils.c ../bluez-5.29/btio/btio.c ../bluez-5.29/src/log.c `pkg-config glib-2.0 --libs`
bluepy-helper.c:25:20: fatal error: config.h: No such file or directory
 #include "config.h"
                    ^
compilation terminated.
../bluez-5.29/lib/bluetooth.c:27:20: fatal error: config.h: No such file or directory
 #include <config.h>
                    ^
compilation terminated.
../bluez-5.29/lib/hci.c:27:20: fatal error: config.h: No such file or directory
 #include <config.h>
                    ^
compilation terminated.
../bluez-5.29/lib/sdp.c:28:20: fatal error: config.h: No such file or directory
 #include <config.h>
```

IanHarvey, 2015-06-12:
I have an update to 5.29 which I'm hoping to push soon. If you're interested in testing it, I'll push it as a branch first before merging to 'master'.

IanHarvey, 2015-06-14:
This is now available as branch **testing/bluez-5.29**  - this has received basic testing with a very limited range of platforms. I'd be very happy if you (or anyone else reading) would be willing to try it in their system.

IanHarvey, 2015-09-02:
I've now merged **testing/bluez-5.29** to master, so it's now officially the supported version.

Any feedback on this change would be welcome; I hope this may fix bugs in communicating with more modern devices.

closed 2015-09-02 by IanHarvey