IanHarvey/bluepy #42: Subscribing to notifications in the GATT client

JPMJPM, 2015-04-22:
Hi,

Could someone tell me how the GATT client has to subscribe to notifications ?. Writting 0x01 in the characteristic CCCD descriptor ?.

I don't understand the way to do this described in bluepy 0.9.0 documentation (working with notifications).

Thanks and regards.

geraudfusar, 2015-05-27:
I currently have the same issue. I am able to connect to my device and set the levels of the digital outputs, but the notifications are not properly fired.

IanHarvey, 2015-06-11:
The relevant Bluetooth documentation page is this: https://developer.bluetooth.org/gatt/descriptors/Pages/DescriptorViewer.aspx?u=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml

You need to find the handle for the Client Characteristic Configuration register and write a 2 byte value to it (01 00 to enable, 00 00 to disable). The lines in SensorTag.py which do this are:

```python
    def enable(self):
        self.periph.writeCharacteristic(0x60, struct.pack('<bb', 0x01, 0x00))

    def disable(self):
        self.periph.writeCharacteristic(0x60, struct.pack('<bb', 0x00, 0x00))
```

where 0x60 is the handle for the Keypress characteristic configuration.

JPMJPM, 2015-06-11:
Ian, thanks for the information.

closed 2015-09-02 by IanHarvey