IanHarvey/bluepy #44: Support for Mac OS X (10.9, 10.10)

vielmetti, 2015-04-29:
bluepy and bluepy-helper depend on Linux's bluez. 

It would be a useful thing to have an abstraction layer that let you write the same python code for both Mac OS X and Linux, along the lines of what noble does for node.js.

Tracking any specific issues here.

wind-rider, 2016-03-01:
https://bitbucket.org/OscarAcena/pygattlib does this, you can have a look at how they integrated the OSX part and the abstraction layer there. I would also say that bluepy could improve upon this point - being cross platform would be a major advantage.

mritchie712, 2016-04-10:
@wind-rider are you syaing pygattlib supports OSX? When I try to install it on OSX I get:

    OSError: Not supported OS

wind-rider, 2016-04-11:
@mritchie712 I'm sorry, I must have confused it.

However here https://github.com/matthewelse/bleep/ says to have experimental OSX support besides Linux support, with two backends as shown in https://github.com/matthewelse/bleep/blob/master/bleep/backend.py.

This could be an example for bluepy?

mritchie712, 2016-04-12:
@wind-rider that looks promising, thanks!

bretcj7, 2017-12-19:
@mritchie712  did you get past the issue?  I'm battling the same right now.

mritchie712, 2017-12-19:
No, I ended up using ionic (JavaScript framework) instead

IanHarvey, 2018-07-23:
Realistically, Mac OS support isn't going to be added to Bluepy any time soon. 

Interested parties might want to look at https://github.com/TheCellule/python-bleson, which supports Mac OS but is still somewhat under development.

closed 2018-07-23 by IanHarvey