IanHarvey/bluepy #54: Error in writing a characteristic

JPMJPM, 2015-06-03:
Hi, 

Trying write a characteristic appears the following error :

Unhandled exception in thread started by <function peticion_datos_PTHVcc at 0xb6adc830>
Traceback (most recent call last):
  File "duracion_bateria.py", line 218, in peticion_datos_PTHVcc
    carac_comando.write("58".decode("hex"), True)
  File "/home/pi/python/btle.py", line 142, in write
    self.peripheral.writeCharacteristic(self.valHandle, val, withResponse)
No se encuentra el dispositivo BLE
  File "/home/pi/python/btle.py", line 377, in writeCharacteristic
    return self._getResp('wr')
  File "/home/pi/python/btle.py", line 262, in _getResp
    resp = Peripheral.parseResp(rv)
  File "/home/pi/python/btle.py", line 227, in parseResp
    (tag, tval) = item.split('=')
ValueError: need more than 1 value to unpack

Could someone tell me what I am doing bad ?

Thanks and regards.

IanHarvey, 2015-06-06:
Could you set 'Debugging = True' (line 11 of btle.py) and run the program again. This will print some extra debugging information on the screen. If you can tell me what this is I'll take a look at it.

Thanks
Ian

JPMJPM, 2015-06-06:
Hi Ian, this is the log with 'Debugging = True'. The python script is a loop with write - notification - read - time.sleep(10) - write - notification - ......... The error appear in the third write :

Regards.

[image]

nizamalki, 2015-09-16:
This is my output for the same issue:

Connecting to 1C:BA:8C:20:E2:A8
Running  /home/pi/bluepy/bluepy/bluepy/bluepy-helper
Sent:  conn 1C:BA:8C:20:E2:A8 public

Got: '# main() :bluepy-helper.c built at 10:59:13 on Sep 16 2015\n'
Got: '# main() :mgmt_register(MGMT_EV_DEVICE_CONNECTED) failed\n'
Got: '# main() :Starting loop\n'
Got: '# mgmt_debug() :# mgmt: [0xffff] command 0x0001\n'
Got: '\n'
Traceback (most recent call last):
  File "sensortag.py", line 293, in <module>
    main()
  File "sensortag.py", line 247, in main
    tag = SensorTag(arg.host)
  File "sensortag.py", line 179, in __init__
    Peripheral.__init__(self,addr)
  File "/home/pi/bluepy/bluepy/bluepy/btle.py", line 192, in __init__
    self.connect(deviceAddr, addrType)
  File "/home/pi/bluepy/bluepy/bluepy/btle.py", line 320, in connect
    rsp = self._getResp('stat')
  File "/home/pi/bluepy/bluepy/bluepy/btle.py", line 277, in _getResp
    resp = Peripheral.parseResp(rv)
  File "/home/pi/bluepy/bluepy/bluepy/btle.py", line 240, in parseResp
    (tag, tval) = item.split('=')
ValueError: need more than 1 value to unpack
Sent:  disc

Got: '# mgmt_debug() :# mgmt: < 01 00 ff ff 00 00                                ......          \n'
Got: '\n'
Exception ValueError: 'need more than 1 value to unpack' in <bound method SensorTag.__del__ of <__main__.SensorTag instance at 0xdb1f30>> ignored

Any help much appreciated!

IanHarvey, 2017-02-06:
It looks like this got fixed in 693d455 (and similar issues in 79adf50), in Sept 2015, but I never closed this issue.

closed 2017-02-06 by IanHarvey