IanHarvey/bluepy #55: Question about connection timeout

JPMJPM, 2015-06-07:
Hi,

What is the default connection timeout value ?
Is it possible to change it ?

Thanks and regards.

Counterfeiter, 2015-08-04:
You could loop and quit the script with bash script
timeout 10 python sensortag.py ....

JPMJPM, 2015-08-05:
Thanks, I will  try it.

akorevaar, 2015-10-13:
Another way of doing it would be to add a timeout to the connect function (see btle.py)

    def connect(self, addr, addrType, timeout=None):
        ...
        rsp = self._getResp('stat', timeout)
        ...

celosiag, 2018-05-10:
Hi,

Sorry to open again this thread so late but @akorevaar can you explicit a bit on your thoughts to implement a connection timeout ? 

I'm interested to set a timeout in connect function but my tries failed. Indeed, when I called the connect function with a setted timeout as you previously describe, I got the following error message : "NoneType' object has no attribute '__getitem__'". 

Thanks in advance for your help :)

mdxs, 2020-10-21:
Resolved with #374 (and while a merge is pending, you could use a MyPeripheral with the PR changes).