IanHarvey/bluepy #56: Device disconnects when running getServices() but works fine with getCharacteristics()

swbooking, 2015-06-09:
Hey Ian, 

I'm having an issue when trying get or discover services. I can establish the connection without an issue:

    >>> conn = Peripheral("5C:F3:70:62:FE:5D", "public")
    Running  /home/pi/GitHub/bluepy/bluepy/bluepy-helper
    Sent:  conn 5C:F3:70:62:FE:5D public

    Got: '# bluepy-helper.c built at 22:58:35 on Jun  8 2015\n'
    Got: "rsp=$stat state=$tryconn dst='5C:F3:70:62:FE:5D mtu=h0 sec='low\n"
    Got: "rsp=$stat state=$conn dst='5C:F3:70:62:FE:5D mtu=h0 sec='low\n"'

Then calling getServices() or discoverServices() causes the device to disconnect:

    >>> conn.discoverServices()
    Sent:  svcs

    Got: "rsp=$stat state=$disc mtu=h0 sec='low\n"
    Stopping  /home/pi/GitHub/bluepy/bluepy/bluepy-helper
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
      File "btle.py", line 325, in discoverServices
        rsp = self._getResp('find')
      File "btle.py", line 286, in _getResp
        raise BTLEException(BTLEException.DISCONNECTED, "Device disconnected")
    btle.BTLEException: Device disconnected

But calling getCharacteristics() works fine:
     
     >>> conn.getCharacteristics()
     Sent:  char 1 FFFF

    Got: "rsp=$find hnd=h4 props=h2 vhnd=h6 uuid='00002a00-0000-1000-8000-00805f9b34fb hnd=h7     props=h2 vhnd=h8 uuid='00002a01-0000-1000-8000-00805f9b34fb hnd=hFFFD props=h1A   vhnd=hFFFE uuid='faf69fd8-d058-45e3-a06e-86662c8c7918\n"
    [<btle.Characteristic instance at 0x76abce40>, <btle.Characteristic instance at 0x76abe0a8>,     <btle.Characteristic instance at 0x76abe170>]

And calling getDescriptors works (minus it seeming to hang at the end):

     >>> conn.getDescriptors()
    Sent:  desc 1 FFFF

    Got: "rsp=$desc hnd=h1 uuid='2800 hnd=h4 uuid='2803 hnd=h6 uuid='2a00 hnd=h7 uuid='2803 hnd=h8 uuid='2a01\n"
    Got: "rsp=$desc hnd=h10 uuid='2800 hnd=hFFFC uuid='2800 hnd=hFFFD uuid='2803\n"
    Got: "rsp=$desc hnd=hFFFE uuid='faf69fd8-d058-45e3-a06e-86662c8c7918\n"
    Got: "rsp=$desc hnd=hFFFF uuid='2902\n"

Any ideas on why the device disconnects? Thanks!

swbooking, 2015-06-10:
The device I am connecting to looks like its using BlueZ-5.24 (raspberry pi). From what I've managed to track through it looks like /attrib/gatt.c is fairly different in both these versions. Could this be where the issue stems from?

IanHarvey, 2015-06-12:
There's no requirement for the versions of BlueZ to be the same on both sides, as the GATT commands being used are (or should be) entirely 'standard'.

Do you have the source code to the peripheral end running on the Raspberry Pi, or are there log files you can see? My best guess is that something is going wrong at that end, in the code that handles 'discoverServices()', and the Pi's BlueZ stack is dropping the connection as a result.

IanHarvey, 2016-03-26:
I'm closing this for now. For what it's worth, I have tried to get some bluez's BLE peripheral code to work on a Pi, and it doesn't work for me either. The problem seems to be on the device code on the Pi, and not at the bluepy end.

closed 2016-03-26 by IanHarvey