IanHarvey/bluepy #57: Connect fails immediately if another connect is pending.

jef-pearlman, 2015-06-16:
I realize I may be pushing the limits of how this is intended to work, and am using a modified bluepy, but here goes:  The short version is that if a bluepy connection is pending (e.g., the receiver is powered off and we're waiting for it to time out), other connection attempts--in separate btle objects, with their own bluepy-helper processes--immediately fail.

I'm using a single python application to communicate with multiple BLE devices.  Now normally you can't get into the state I described above (multiple pending connects) because connect() blocks.  You might be able to get there by using threads, but for various reasons I took a different approach, and made my own asynchronous connect method, which initiates the connect, and a second method that you can poll for when the connect completes (either successfully or not).  Aside from the problem described here, this appears to work perfectly.  If all receivers are up, or if I wait for the first connection to fail before initiating the second connection, then they all connect and operate simultaneously as expected.

So in my code, the steps that cause this are:
1. Instantiate btle object btleA.
2. Instantiate btle object btleB.
3. Call btleA.connect_async()
4. Call btleB.connect_async()
5. Immediate failure on #4.
6. If you wait until btleA connection times out and call btleB.connect_async() again, it connects  fine.

When it fails this way, it prints the following message:
```connect: Device or resource busy (16)```

I believe I have tracked this to the connect() call in bluez btio.c line 360, but I don't know enough about bluez to really dig further.  
 
FWIW, I just tried it on the testing/bluez-5.29 branch, and it behaved exactly the same.

Thanks for any help, and let me know if you need additional info or if there's a better way I should be doing all of this; I realize this is a bit scattered.  I'm also happy to share my async connection code, though it isn't robust and I'm 90% it's not the direct cause of the issue.

jef-pearlman, 2015-06-16:
I just tried creating the same situation by running two separate gatttool processes, and had the same result: while one connect is pending, others immediately fail.  So presumably this is not a bluepy issue, but something lower level.  I'd still love any suggestions on workarounds, fixes, or working-as-intended.  Thanks!

IanHarvey, 2015-07-07:
It looks like it's a limitation of Bluetooth LE itself. Connecting to a peripheral is done with the **LE_Create_Connection** command; the Bluetooth 4.0 spec says (vol 2, sec 7.8.12, page 826)

> The Host shall not issue this command when another LE_Create_Connection is pending in the
> Controller; if this does occur the Controller shall return the Command Disallowed error code shall 
> be used.

Sadly, then, it's built in to the HCI command specification that only one connection can be pending at a a time.

Thanks
Ian

closed 2015-07-07 by IanHarvey

jef-pearlman, 2015-07-07:
Apologies--I meant to close this myself.  In case it is ever of use to anyone else, I also tracked down the kernel code implementing the limitation, and it appears to only limit it to one connection per CID (rather than one globally).  Since all my devices are using the Attribute CID (0x04) and I don't really have control over it, I haven't tested whether you can do multiple connections on different channels or the limitation matches the BT4 spec, though I note that the error returned is EBUSY.  Thanks!