IanHarvey/bluepy #70: raw BLE advertisement scanning

msnyder415, 2015-09-08:
Hi Ian 

```
- Implement 'hcitool lescan' functionality
```

Are you aware of any python libs that will expose the raw advertisement payloads? 
Any idea if / when yours may? 

Thanks!

ShuffleBox, 2015-09-08:
I've used PyGattlib for BLE device discovery. https://bitbucket.org/OscarAcena/pygattlib
Perhaps worth a look.

IanHarvey, 2015-12-03:
Scanning is now available (albeit there are no docs yet) in v/0.9.10, just out.

The raw advertisement payloads come over as the _rawData_ attribute of a _ScanEntry_ object, which is passed to the delegate registered with a _Scanner_ object. There's now a program **test/scan.py** which should give you the basic idea.

Thanks
Ian

closed 2015-12-03 by IanHarvey

NateEaton, 2021-04-14:
I know this is a stale topic but I need the same thing, access to the raw advertisement data. I don't see any ref to rawData in the 0.9.11 docs... was it dropped? Or just still not documented?

B03901108, 2022-05-25:
I share similar feedback with @NateEaton that some class members like rawData and scanData are not available in official documents. I had to find them by diving into the source code (btle.py). Just for the convenience of others working on low-level BLE stuff, rawData provides the raw payload of the latest observed advertisement, while scanData provides the accumulated AD Data keyed by AD Type.

Say, if the current scan request invites no scan responses, but an earlier scan request invites a scan response, rawBinary implies the lack of scan response, while scanData still keeps some information from the earlier scan response.