IanHarvey/bluepy #72: btle.py doesnt work after reboot

shailurie, 2015-09-16:
I've installed bluez 5.30 (downloaded and built manually).
Then clone and build bluepy.
Then I ran python btle.py XX:YY:...... and it worked perfectly.

However after the first restart is doesnt work anymore.

Connecting to: C4:BE:84:76:F3:14, address type: public
Traceback (most recent call last):
  File "btle.py", line 481, in <module>
    conn = Peripheral(devAddr, addrType)
  File "btle.py", line 193, in __init__
    self.connect(deviceAddr, addrType)
  File "btle.py", line 321, in connect
    rsp = self._getResp('stat')
  File "btle.py", line 278, in _getResp
    resp = Peripheral.parseResp(rv)
  File "btle.py", line 241, in parseResp
    (tag, tval) = item.split('=')
ValueError: need more than 1 value to unpack
Exception ValueError: 'need more than 1 value to unpack' in <bound method Peripheral.__del__ of <__main__.Peripheral instance at 0x76aef4e0>> ignored

shailurie, 2015-09-21:
It seems like something in the one of the last changes caused the bluepy to stop working after restart.
the workaround which works for me to comment out the mgmt_debug function (the one above the main function) and the call to the function (in main function).

IanHarvey, 2015-09-24:
I can't reproduce your exact error, but I've spotted a problem in the debugging output which might be the cause. If you could let me know whether commit 693d455 fixes the problem that would be useful.

Thanks
Ian

shailurie, 2015-09-24:
No problem. I will test it later today and let you know.
BTW, what is the latest bluez version (or the recommended ) it can work
with?

Shai

IanHarvey, 2015-09-30:
The "unused variable" warnings should have been fixed in commit 4611544. I found another possible problem with the debugging output, which was fixed in 79adf50.

Let me know if you still have problems.

Thanks,
Ian

closed 2015-09-30 by IanHarvey

shailurie, 2015-09-30:
As mentioned before, you can comment out the content of mgmt_debug and keep
the debugging. I checked it and it works.
Which bluez version should I use with it? (5.29 - 5.35)