IanHarvey/bluepy #74: Registration to notification

pgpexp, 2015-09-27:
Hi I am writing a program that register for a notification, I register the delegate and after  try to register for notification calling write('\x01\x00') as noted in another issue and described in the bluetooth 4.0 rfc.
But it doesn't work, any notification for the characteristic is received. Is there an error in the small snippet I wrote? Thank you so much.

   class MyDelegate(btle.DefaultDelegate):

        def __init__(self, hndl):
            btle.DefaultDelegate.__init__(self)
	   self.hndl=hndl;

       def handleNotification(self, cHandle, data):
	   if (cHandle==self.hndl):
        	    val = binascii.b2a_hex(data)
        		val = binascii.unhexlify(val)
        		val = struct.unpack('f', val)[0]
         		print str(val) + " deg C"

    p = btle.Peripheral("xx:xx:xx:xx", "random")

    try:
	   srvs = (p.getServices());
           chs=srvs[2].getCharacteristics();
          ch=chs[1];
	  print(str(ch)+str(ch.propertiesToString()));
	  p.setDelegate(MyDelegate(ch.getHandle()));
          # Setup to turn notifications on, e.g.
	  ch.write("\x01\x00");

    # Main loop --------
	  while True:
	      if p.waitForNotifications(1.0):
		  continue

	    print "Waiting..."
	    # Perhaps do something else here
     finally:
	    p.disconnect();

pcborenstein, 2015-12-03:
You are trying to write 1 to the characteristic itself. The notification enable is always the handle after the target characteristic.

Do this:
```
ConfigHndl = ch.valHandle +1
p.writeCharacteristic(ConfigHndl , (1).to_bytes(2, byteorder='little'))
```

IanHarvey, 2016-01-20:
Did this fix the problem?

Also, it's not generally reliable to pick out indices into lists of values ( *srvs[2]*, *chs[1]* etc.) because bluepy doesn't guarantee the order (see e.g. http://stackoverflow.com/questions/526125/why-is-python-ordering-my-dictionary-like-so for some of the background to this).

In general, you either have to find the services / characteristics you want by UUID, or consult the documentation for the BLE peripheral to discover the handles for the characteristics you're trying to read or write.

Cheers
Ian

closed 2017-06-26 by IanHarvey