IanHarvey/bluepy #78: Upgrade to BlueZ-5.35

rmoshiri, 2015-10-05:
I tried to replace the bluez-5.29 in this package, with the latest bluez-5.35 release.
Got compile error in "bluepy-helper.c", line 462. Made minor change there, and it 
seems to work. I just set the new flag passed to the subroutine as "false". That
should be OK as it is for an external encryption option.

Just in case you want to upgrade and test it, below were the chaanges:

sudo leafpad Makefile         (Change Bluez version's Directory)
                                BLUEZ_PATH=../bluez-5.35
 sudo leafpad bluepy-helper.c  (Add "false" for a 3rd/new parameter RE external encryption)
                                Line 462: attrib = g_attrib_new(iochannel, mtu, false);

raplin, 2015-12-29:
FYI This fix is required & works when building against bluez 5.37 (latest at time of writing) also. Thanks!

IanHarvey, 2018-09-22:
bluepy 1.2.0 now uses code from bluez 5.47, so I'll close this issue now.

closed 2018-09-22 by IanHarvey