IanHarvey/bluepy #82: Can't Read Characteristic

james-bui, 2015-11-10:
Hi,

I have five characteristics on my bluetooth sensor but when I loop through the list of characteristics to be read, I end up only reading the first characteristic, which is the name of my sensor. When it hits the second characteristic, it just stalls there and never reads. When I cancel out of it, I get the error: "Exception IOError: (32, 'Broken pipe') in <bound method Peripheral.__del__ of <btle.Peripheral instance at 0x76c423c8>> ignored". I don't really know why this is happening. Is this an issue with how the read function is supposed to work? Thanks.

IanHarvey, 2016-01-20:
I can't really help without more information here.
- What is the sensor you're trying to read from? If it's a SensorTag, does **sensortag.py** work?
- If you run **python btle.py <MAC address>** it should attempt to list all characteristics, and read from those which indicate that they support the read operation. Does this work?
- If you set *Debugging = True* at the top of **btle.py**, what does the debug output say?

Thanks
Ian

james-bui, 2016-01-22:
Thanks for answering my question Ian. I'm currently using a Bluegiga BLE112 smart module. 

When I run python btle.py, it reads through my first characteristic but then stalls when it comes the next once again. I get this:

Connecting to: 00:07:80:60:D6:E7, address type: public
Service <uuid=Generic Access handleStart=1 handleEnd=3> :
    Characteristic <Device Name>, supports READ 
    -> 'SmartDerm V3.0'
Service <uuid=203f handleStart=4 handleEnd=65535> :
    Characteristic <0001>, supports READ 

It stalls until I Ctrl-C it, in which case it gives me the following: 

^CTraceback (most recent call last):
  File "btle.py", line 496, in <module>
    conn.disconnect()
  File "btle.py", line 337, in disconnect
    self._getResp('stat')
  File "btle.py", line 274, in _getResp
    rv = self._helper.stdout.readline()
IOError: [Errno 0] Error

With Debugging = True, I get the following:

Connecting to: 00:07:80:60:D6:E7, address type: public
Running  /home/pi/bluepy/bluepy/bluepy-helper
Sent:  conn 00:07:80:60:D6:E7 public

Got: "rsp=$stat state=$tryconn dst='00:07:80:60:D6:E7 mtu=h0 sec='low\n"
Got: "rsp=$stat state=$conn dst='00:07:80:60:D6:E7 mtu=h0 sec='low\n"
Sent:  svcs

Got: "rsp=$find hstart=h1 hend=h3 uuid='00001800-0000-1000-8000-00805f9b34fb hstart=h4 hend=hFFFF uuid='0000203f-0000-1000-8000-00805f9b34fb\n"
Service <uuid=Generic Access handleStart=1 handleEnd=3> :
Sent:  char 1 3

Got: "rsp=$find hnd=h2 props=h2 vhnd=h3 uuid='00002a00-0000-1000-8000-00805f9b34fb\n"
    Characteristic <Device Name>, supports READ 
Sent:  rd 3

Got: 'rsp=$rd d=b536D6172744465726D2056332E30\n'
    -> 'SmartDerm V3.0'
Service <uuid=203f handleStart=4 handleEnd=65535> :
Sent:  char 4 FFFF

Got: "rsp=$find hnd=h5 props=h2 vhnd=h6 uuid='00000001-0000-1000-8000-00805f9b34fb hnd=h8 props=h20 vhnd=h9 uuid='00000002-0000-1000-8000-00805f9b34fb hnd=hC props=h20 vhnd=hD uuid='00000003-0000-1000-8000-00805f9b34fb hnd=h10 props=h20 vhnd=h11 uuid='00000009-0000-1000-8000-00805f9b34fb\n"
    Characteristic <0001>, supports READ 
Sent:  rd 6

But once again, it just stalls here and nothing happens until I Ctrl-C, in which it gives me a traceback: 

^CSent:  disc

Traceback (most recent call last):
  File "btle.py", line 496, in <module>
    conn.disconnect()
  File "btle.py", line 337, in disconnect
    self._getResp('stat')
  File "btle.py", line 274, in _getResp
    rv = self._helper.stdout.readline()
IOError: [Errno 0] Error

I have been stuck in trying to get it to read the next 4 characteristics as those are the characteristics in which I get my pressure data from but I have no idea how to resolve this problem. Any help would be greatly appreciated.