IanHarvey/bluepy #83: cannot connect using btle.py

10hsin, 2015-11-16:
Hi Ian, I've met some problems when trying to connect to BLE device via btle.py
I can have a success connection using gatttool 
But failed by using btle.py, both public mode and random mode

Below are error messages:
-------------------------------------------public mode-----------------------------------------------------
Connecting to: F3:12:55:AE:64:11, address type: public
Traceback (most recent call last):
  File "btle.py", line 482, in <module>
    conn = Peripheral(devAddr, addrType)
  File "btle.py", line 194, in __init__
    self.connect(deviceAddr, addrType)
  File "btle.py", line 328, in connect
    "Failed to connect to peripheral %s, addr type: %s" % (addr, addrType))
__main__.BTLEException: Failed to connect to peripheral F3:12:55:AE:64:11, addr type: public

-------------------------------------------random mode-----------------------------------------------------
Connecting to: F3:12:55:AE:64:11, address type: random
Service <uuid=Generic Attribute handleStart=8 handleEnd=8> :
Traceback (most recent call last):
  File "btle.py", line 486, in <module>
    for ch in svc.getCharacteristics():
  File "btle.py", line 103, in getCharacteristics
    self.chars = self.peripheral.getCharacteristics(self.hndStart, self.hndEnd)
  File "btle.py", line 379, in getCharacteristics
    rsp = self._getResp('find')
  File "btle.py", line 306, in _getResp
    raise BTLEException(BTLEException.COMM_ERROR, "Error from Bluetooth stack (%s)" % errcode)
__main__.BTLEException: Error from Bluetooth stack (comerr)

The device I use is CSR 4.0 dongle on RPi and Red Bear BLE Shiled
I use Debian Wheezy and Bluez 5.29
I've tried to modify btle.py discussed in issue 27, but still not working
Any suggestion?
Thanks for your contribution

IanHarvey, 2015-11-18:
Could you try opening a second terminal window, and running

    sudo hcidump -t

in it, and then try running btle.py in the first window. This should give lots of debug output - if you can post this I'll take a look at it. It would also be useful if you could post the output when you connect with gatttool.

If *hcidump* isn't found, you might need to install it with 

    sudo apt-get install bluez-hcidump

Thanks
Ian

10hsin, 2015-11-23:
Hello Ian, 
    Below are the output of gatttool connection using hcidump 
    hcidump - gatttool
        < HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2
        0000: 46 00                                             F.
        > HCI Event: Command Status (0x0f) plen 4
        LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
        > HCI Event: LE Meta Event (0x3e) plen 12
        LE Read Remote Used Features Complete
        status 0x00 handle 70
        Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00

    But no output for sudo python btle.py F3:12:55:AE:64:11 random

10hsin, 2015-11-23:
Hello Ian, below are some output of hcidump

-------------------------------------------------------------------------
gatttool connection
-------------------------------------------------------------------------
	
> 	HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2
	0000: 46 00                                             F.
> 	HCI Event: Command Status (0x0f) plen 4
        LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
> 	HCI Event: LE Meta Event (0x3e) plen 12
	LE Read Remote Used Features Complete
	status 0x00 handle 70
	Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00

-------------------------------------------------------------------------
btle public mode
-------------------------------------------------------------------------
>	HCI Command: LE Create Connection Cancel (0x08|0x000e) plen 0
> 	HCI Event: Command Complete (0x0e) plen 4
	LE Create Connection Cancel (0x08|0x000e) ncmd 1
	0000: 00                                                .
> 	HCI Event: LE Meta Event (0x3e) plen 19
	LE Connection Complete
	status 0x02 handle 0, role master
	bdaddr F3:12:55:AE:64:11 (Public)
< 	HCI Command: LE Create Connection Cancel (0x08|0x000e) plen 0

-------------------------------------------------------------------------
btle random mode
-------------------------------------------------------------------------

> 	HCI Event: Command Status (0x0f) plen 4
	Unknown (0x00|0x0000) status 0x00 ncmd 1
> 	HCI Event: LE Meta Event (0x3e) plen 12
	LE Read Remote Used Features Complete
	status 0x00 handle 70
	Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
< 	ACL data: handle 70 flags 0x00 dlen 11
    ATT: Read By Group req (0x10)
	start 0x0001, end 0xffff
	type-uuid 0x2800
> 	HCI Event: Number of Completed Packets (0x13) plen 5
	handle 70 packets 1
> 	ACL data: handle 70 flags 0x02 dlen 18
	ATT: Read By Group resp (0x11)
	attr handle 0x0001, end group handle 0x0007
	value 0x00 0x18
	attr handle 0x0008, end group handle 0x0008
	value 0x01 0x18
< 	ACL data: handle 70 flags 0x00 dlen 11
	ATT: Read By Group req (0x10)
	start 0x0009, end 0xffff
	type-uuid 0x2800
> 	HCI Event: Number of Completed Packets (0x13) plen 5
	handle 70 packets 1
> 	ACL data: handle 70 flags 0x02 dlen 26
	ATT: Read By Group resp (0x11)
	attr handle 0x0009, end group handle 0x000e
	value 0x1e 0x94 0x8d 0xf1 0x48 0x31 0x94 0xba 0x75 0x4c 0x3e 0x50 0x00 0x00 0x3d 0x71
< 	ACL data: handle 70 flags 0x00 dlen 11
	ATT: Read By Group req (0x10)
	start 0x000f, end 0xffff
	type-uuid 0x2800
> 	HCI Event: Number of Completed Packets (0x13) plen 5
	handle 70 packets 1
> 	ACL data: handle 70 flags 0x02 dlen 12
	ATT: Read By Group resp (0x11)
	attr handle 0x000f, end group handle 0xffff
	value 0x0a 0x18
< 	ACL data: handle 70 flags 0x00 dlen 11
	ATT: Read By Type req (0x08)
	start 0x0008, end 0x0008
	type-uuid 0x2803
> 	HCI Event: Number of Completed Packets (0x13) plen 5
	handle 70 packets 1
> 	ACL data: handle 70 flags 0x02 dlen 9
	ATT: Error (0x01)
	Error: Attribute not found (10)
	Read By Type req (0x08) on handle 0x0008
< 	HCI Command: Disconnect (0x01|0x0006) plen 3
	handle 70 reason 0x13
	Reason: Remote User Terminated Connection
> 	HCI Event: Command Status (0x0f) plen 4
	Disconnect (0x01|0x0006) status 0x00 ncmd 1
> 	HCI Event: Disconn Complete (0x05) plen 4
	status 0x00 handle 70 reason 0x16
	Reason: Connection Terminated by Local Host

s-t-e-f-a-n, 2015-12-09:
I have a similar problem on my Synology DS211+ NAS when trying to put a CC2650 into operation - hcitool lescan and lecc works - also the scan.py:
```javascript
python scan.py
Scanning for devices...
    Device (new): b0:b4:48:be:d1:01 (public), -82 dBm
        Flags: <05>
        Incomplete 16b Services: <80aa>
        Manufacturer: <0d00030000>
    Device (update): b0:b4:48:be:d1:01 (public), -82 dBm
        Flags: <05>
        Incomplete 16b Services: <80aa>
        Complete Local Name: 'CC2650 SensorTag'
        Tx Power: <00>
        0x12: <08002003>
        Manufacturer: <0d00030000>
```

Here is the errornoues output of btle.py (all debugs switched on):
```javascript
python btle.py B0:B4:48:BE:D1:01
Connecting to: B0:B4:48:BE:D1:01, address type: public
Running  /volume1/EarthG/projects/SensorTag/bluepy/bluepy/bluepy-helper
Sent:  conn B0:B4:48:BE:D1:01 public

Got: '# main() :bluepy-helper.c built at 00:20:29 on Dec  9 2015\n'
Got: '# main() :Starting loop\n'
Got: "rsp=$stat state=$tryconn dst='B0:B4:48:BE:D1:01 mtu=h0 sec='low\n"
Got: '# cmd_connect() :gatt_connect returned 0x36ef8\n'
Got: '# connect_cb() :io = 0x36ef8, err = (nil)\n'
Got: "# Can't detect MTU, using defaultrsp=$stat state=$conn dst='B0:B4:48:BE:D1:01 mtu=h0 sec='low\n"
Got: '# channel_watcher() :chan = 0x36ef8\n'
Got: "rsp=$stat state=$disc mtu=h0 sec='low\n"
Stopping  /volume1/EarthG/projects/SensorTag/bluepy/bluepy/bluepy-helper
Traceback (most recent call last):
  File "btle.py", line 683, in <module>
    conn = Peripheral(devAddr, addrType)
  File "btle.py", line 323, in __init__
    self.connect(addr, self.addrType, self.iface)
  File "btle.py", line 370, in connect
    "Failed to connect to peripheral %s, addr type: %s" % (addr, addrType))
__main__.BTLEException: Failed to connect to peripheral B0:B4:48:BE:D1:01, addr type: public
```
Sometimes there is no traceback and it runs without further output, and sometimes the traceback is output immediately.
The problem occurrs with python 2.7 as well as with 3.2.
hcidump is not available easily - ipkg-bootstrap does not offer it....It was already hard for me to get the bluepy-helper to be compiled on the NAS...

Inachis, 2016-02-06:
Hi Ian,
I have similar problem. I can connect with gatttool and hcitool lecc.

**python btle.py  98:4F:EE:0D:06:1A**

Connecting to: 98:4F:EE:0D:06:1A, address type: public
Service <uuid=Generic Attribute handleStart=8 handleEnd=8> :
Traceback (most recent call last):
  File "btle.py", line 687, in <module>
    for ch in svc.getCharacteristics():
  File "btle.py", line 106, in getCharacteristics
    self.chars = self.peripheral.getCharacteristics(self.hndStart, self.hndEnd)
  File "btle.py", line 420, in getCharacteristics
    rsp = self._getResp('find')
  File "btle.py", line 334, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "btle.py", line 296, in _waitResp
    raise BTLEException(BTLEException.COMM_ERROR, "Error from Bluetooth stack (%s)" % errcode)
__main__.BTLEException: Error from Bluetooth stack (comerr)

**hcidump -t**:

HCI sniffer - Bluetooth packet analyzer ver 5.37
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-02-06 21:59:47.963605 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 98:4F:EE:0D:06:1A type 0
    interval 96 window 48 initiator_filter 0
    own_bdaddr_type 0 min_interval 40 max_interval 56
    latency 0 supervision_to 42 min_ce 0 max_ce 0
2016-02-06 21:59:47.969782 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2016-02-06 21:59:47.995735 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 71, role master
      bdaddr 98:4F:EE:0D:06:1A (Public)
2016-02-06 21:59:47.996304 < HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2
2016-02-06 21:59:48.001769 > HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 0
2016-02-06 21:59:48.239780 > HCI Event: Command Status (0x0f) plen 4
    Unknown (0x00|0x0000) status 0x00 ncmd 1
2016-02-06 21:59:48.379800 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Read Remote Used Features Complete
      status 0x00 handle 71
      Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
2016-02-06 21:59:48.399172 < ACL data: handle 71 flags 0x00 dlen 11
    ATT: Read By Group req (0x10)
      start 0x0001, end 0xffff
      type-uuid 0x2800
2016-02-06 21:59:48.449816 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 71 packets 1
2016-02-06 21:59:48.590232 > ACL data: handle 71 flags 0x02 dlen 24
    ATT: Read By Group resp (0x11)
      attr handle 0x0001, end group handle 0x0007
      value 0x00 0x18
      attr handle 0x0008, end group handle 0x0008
      value 0x01 0x18
      attr handle 0x0009, end group handle 0xffff
      value 0x0d 0x18
2016-02-06 21:59:48.601999 < ACL data: handle 71 flags 0x00 dlen 11
    ATT: Read By Type req (0x08)
      start 0x0008, end 0x0008
      type-uuid 0x2803
2016-02-06 21:59:48.659853 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 71 packets 1
2016-02-06 21:59:48.729112 > ACL data: handle 71 flags 0x02 dlen 9
    ATT: Error (0x01)
      Error: Attribute not found (10)
      Read By Type req (0x08) on handle 0x0008
2016-02-06 21:59:50.729659 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 71 reason 0x13
    Reason: Remote User Terminated Connection
2016-02-06 21:59:50.731086 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2016-02-06 21:59:50.760132 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 71 reason 0x16
    Reason: Connection Terminated by Local Host

If it helps, here's a successful gatttool output
**gatttool -b 98:4F:EE:0D:06:1A -I**

[98:4F:EE:0D:06:1A][LE]> connect
Attempting to connect to 98:4F:EE:0D:06:1A
Connection successful
[98:4F:EE:0D:06:1A][LE]> primary
attr handle: 0x0001, end grp handle: 0x0007 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle: 0x0008, end grp handle: 0x0008 uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle: 0x0009, end grp handle: 0xffff uuid: 0000180d-0000-1000-8000-00805f9b34fb
[98:4F:EE:0D:06:1A][LE]> char-desc 0x0009 0xffff
handle: 0x0009, uuid: 00002800-0000-1000-8000-00805f9b34fb
handle: 0x000a, uuid: 00002803-0000-1000-8000-00805f9b34fb
handle: 0x000b, uuid: 00002a37-0000-1000-8000-00805f9b34fb
handle: 0x000c, uuid: 00002902-0000-1000-8000-00805f9b34fb
[98:4F:EE:0D:06:1A][LE]> char-write-req 0x000b 0100
Error: Characteristic Write Request failed: Attribute can't be written
[98:4F:EE:0D:06:1A][LE]> char-write-req 0x000c 0100
Characteristic value was written successfully
Notification handle = 0x000b value: 00 4a
Notification handle = 0x000b value: 00 4b
Notification handle = 0x000b value: 00 4a
Notification handle = 0x000b value: 00 4b
Notification handle = 0x000b value: 00 4a
Notification handle = 0x000b value: 00 4b
Notification handle = 0x000b value: 00 4a
Notification handle = 0x000b value: 00 4b
Notification handle = 0x000b value: 00 4a
Notification handle = 0x000b value: 00 4b
Notification handle = 0x000b value: 00 4a
[98:4F:EE:0D:06:1A][LE]> char-write-req 0x000c 0000
Characteristic value was written successfully
[98:4F:EE:0D:06:1A][LE]>

Any suggestions?
Kelvin

timcknowles, 2016-03-19:
Hi Ian,

Thanks for the code and the really useful eLinux article. I have been following your instructions but I get the following error. I'm trying to connect to an Anova https://github.com/erikcw/pycirculate

Any help would be greatly appreciated. 

Thanks in advance 

 Tim

pi@raspberrypi ~/bluepy/bluepy $ sudo hcitool lecc F4:B8:5E:AF:F8:D6
Connection handle 64
pi@raspberrypi ~/bluepy/bluepy $ python btle.py F4:B8:5E:AF:F8:D6
Connecting to: F4:B8:5E:AF:F8:D6, address type: public
Traceback (most recent call last):
  File "btle.py", line 683, in <module>
    conn = Peripheral(devAddr, addrType)
  File "btle.py", line 318, in __init__
    self.connect(deviceAddr, addrType, iface)
  File "btle.py", line 367, in connect
    "Failed to connect to peripheral %s, addr type: %s" % (addr, addrType))
__main__.BTLEException: Failed to connect to peripheral F4:B8:5E:AF:F8:D6, addr type: public

**pi@raspberrypi ~ $ sudo hcidump -t**
HCI sniffer - Bluetooth packet analyzer ver 5.32
device: hci0 snap_len: 1500 filter: 0xffffffff
2016-03-19 10:27:43.039080 < HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr F4:B8:5E:AF:F8:D6 type 0
    interval 4 window 4 initiator_filter 0
    own_bdaddr_type 0 min_interval 15 max_interval 15
    latency 0 supervision_to 3200 min_ce 1 max_ce 1
2016-03-19 10:27:43.041158 > HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
2016-03-19 10:27:44.025229 > HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr F4:B8:5E:AF:F8:D6 (Public)
2016-03-19 10:27:44.025851 < HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2
2016-03-19 10:27:44.027273 > HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
2016-03-19 10:27:44.079255 > HCI Event: LE Meta Event (0x3e) plen 12
    LE Read Remote Used Features Complete
      status 0x00 handle 64
      Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
2016-03-19 10:27:44.623595 > ACL data: handle 64 flags 0x02 dlen 16
    L2CAP(d): cid 0x0005 len 12 [psm 0]
2016-03-19 10:27:44.623869 < ACL data: handle 64 flags 0x00 dlen 10
    L2CAP(d): cid 0x0005 len 6 [psm 0]
2016-03-19 10:27:44.623954 < HCI Command: LE Connection Update (0x08|0x0013) plen 14
2016-03-19 10:27:44.626317 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
2016-03-19 10:27:44.960353 > HCI Event: LE Meta Event (0x3e) plen 10
    LE Connection Update Complete
      status 0x00 handle 64
      interval 37.50ms, latency 0.00ms, superv. timeout 6000.00ms
2016-03-19 10:27:44.974353 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
2016-03-19 10:27:46.074535 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
2016-03-19 10:27:46.076495 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2016-03-19 10:27:52.048181 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x22
    Reason: LMP Response Timeout

martinegginton, 2016-04-24:
Hi,
I don't know if this is the same issue but I had the same error when using the CC2650 code. I am using **Spyder** to run the python part.
I opened a shell, disabled a usb not related to blue le (*$ sudo hciconfig hci1 down*). Restarted hci1 (up then down) and then re-ran the script.
After that I was able to access the Tag again.
Martin

ElectronicWalks, 2021-07-16:
"Failed to connect to peripheral %s, addr type: %s" % (addr, addrType))
__main__.BTLEException: Failed to connect to peripheral B0:B4:48:BE:D1:01, addr type: public

solutuion:-  sudo python btle.py B0:B4:48:BE:D1:01  random

ElectronicWalks, 2021-07-16:
> I have a similar problem on my Synology DS211+ NAS when trying to put a CC2650 into operation - hcitool lescan and lecc works - also the scan.py:
> 
> ```js
> python scan.py
> Scanning for devices...
>     Device (new): b0:b4:48:be:d1:01 (public), -82 dBm
>         Flags: <05>
>         Incomplete 16b Services: <80aa>
>         Manufacturer: <0d00030000>
>     Device (update): b0:b4:48:be:d1:01 (public), -82 dBm
>         Flags: <05>
>         Incomplete 16b Services: <80aa>
>         Complete Local Name: 'CC2650 SensorTag'
>         Tx Power: <00>
>         0x12: <08002003>
>         Manufacturer: <0d00030000>
> ```
> 
> Here is the errornoues output of btle.py (all debugs switched on):
> 
> ```js
> python btle.py B0:B4:48:BE:D1:01
> Connecting to: B0:B4:48:BE:D1:01, address type: public
> Running  /volume1/EarthG/projects/SensorTag/bluepy/bluepy/bluepy-helper
> Sent:  conn B0:B4:48:BE:D1:01 public
> 
> Got: '# main() :bluepy-helper.c built at 00:20:29 on Dec  9 2015\n'
> Got: '# main() :Starting loop\n'
> Got: "rsp=$stat state=$tryconn dst='B0:B4:48:BE:D1:01 mtu=h0 sec='low\n"
> Got: '# cmd_connect() :gatt_connect returned 0x36ef8\n'
> Got: '# connect_cb() :io = 0x36ef8, err = (nil)\n'
> Got: "# Can't detect MTU, using defaultrsp=$stat state=$conn dst='B0:B4:48:BE:D1:01 mtu=h0 sec='low\n"
> Got: '# channel_watcher() :chan = 0x36ef8\n'
> Got: "rsp=$stat state=$disc mtu=h0 sec='low\n"
> Stopping  /volume1/EarthG/projects/SensorTag/bluepy/bluepy/bluepy-helper
> Traceback (most recent call last):
>   File "btle.py", line 683, in <module>
>     conn = Peripheral(devAddr, addrType)
>   File "btle.py", line 323, in __init__
>     self.connect(addr, self.addrType, self.iface)
>   File "btle.py", line 370, in connect
>     "Failed to connect to peripheral %s, addr type: %s" % (addr, addrType))
> __main__.BTLEException: Failed to connect to peripheral B0:B4:48:BE:D1:01, addr type: public
> ```
> 
> Sometimes there is no traceback and it runs without further output, and sometimes the traceback is output immediately.
> The problem occurrs with python 2.7 as well as with 3.2.
> hcidump is not available easily - ipkg-bootstrap does not offer it....It was already hard for me to get the bluepy-helper to be compiled on the NAS...

solutuion:- sudo python btle.py B0:B4:48:BE:D1:01 random

cypheron, 2021-12-14:
Was getting the same error. It works with ADDR_TYPE_RANDOM!

bwomsm1, 2023-03-15:
Hi @IanHarvey I am also facing this issue, I am running with `sudo` permissions,
Any advice here?