IanHarvey/bluepy #84: Update connection command to include optional interface to use

jspaletavert, 2015-11-19:
I'm working on a situation where I'm going to be connecting to upto 20 devices and I want to spread them across 3 or 4 bluetooth adapters.  bluepy won't let me select which adapter to use yet.

So I did a little digging and I think I can patch it in quite easily.  Here's my plan:
opt_src is always set to NULL in gatt_connect() right now

Instead what I would like is for the conn command to take an additional optional string for the interface to use hci0  hci1  etc...   to set the opt_src in the gatt_conn call.

I'm going to make this an optional argument to the conn command so that the default behavior stays the same.

IanHarvey, 2015-12-03:
I've taken your code and it's now merged on to master with a few bug fixes, along with calandoa's scanning code.

You can now take scan results (the ScanEntry object) and pass it direct to Peripheral's constructor, to make a connection to a device you just scanned. The ScanEntry object keeps a note of the interface being used for the scan, and passes it to Peripheral.connect(), so it ought to work seamlessly.

closed 2015-12-03 by IanHarvey

kg-meso, 2023-02-16:
This important feature seems to be failing, see issue #501