IanHarvey/bluepy #88: Documentation Update

pcborenstein, 2015-12-03:
Would you mind updating the documentation? I've been reading http://ianharvey.github.io/bluepy-doc/peripheral.html. The connect() instance requires 2 arguments, not one as stated in the documentation. The getServices() returns a dictionary, not a list as stated in the documentation. I assume there are more.

How have people been using this package? Is everyone browsing the source code and figuring it out for themselves? Am I looking at the wrong doc?

IanHarvey, 2015-12-03:
Steady on there!  So, the docs haven't been updated since the v/0.9.0 release, and a lot has moved on since then. A lot of people using bluepy are either reading the source, or starting with the example code (sensortag.py) and hacking from there.

To answer your specific questions:
1) Yes, _connect()_ documentation doesn't now match the code. Typically, people aren't calling _connect()_ directly, they're passing a MAC address into the _Peripheral_ constructor, which works as advertised.
2)  _getService()_ should work as described. The code reads:

    def getServices(self):
        if not self.discoveredAllServices:
            self.discoverServices()
        return self.services.values()

This is a proper list on Python 2.x, and a view on 3.x. It's not a dictionary. You should, at least, be able to iterate over it (see the `__main__` section in `btle.py`.

pcborenstein, 2015-12-03:
Hi Ian,

If you think it would help I'll go though and find all the things that don't match. I want to help, but I'm not very experienced...

IanHarvey, 2015-12-03:
Thanks, that would be very helpful. One useful task would be to simply write down which methods have parameters which no longer match the source code - that should be fairly 'mechanical' once you get the hang of it. There is a lot of documentation to write relating to recently added scanning functionality - if you're willing to read through when it's done I'll make sure I let you know.

Cheers
Ian

pcborenstein, 2015-12-08:
In the peripheral class, the connect and init method have the same 3 arguments. The last argument, iface, isn't documented for either. The arguments for all other methods match the documentation.
getServices() returns a view. It seems to be the only thing that returns a veiw.
There are many unlisted methods in the peripheral class. readCharacteristic, addrType, discoverServices, discoveredAllServices, are just a few. 

In the UUID class, the binval method of the UUID class isn't documented. Its very valuable.
Also in the UUID class, the initiation can take a common name which is stashed in the class variable called commonName. You didn't document it, but it seems useless so I wouldn't tell anyone about it.

The service class has 5 unlisted class variables, hndEnd, hndStart, peripheral, uuid, and chars.

The characteristics class has props and propNames which seem to be the same thing

Are these the kinds of things you wanted me to look at?

Side note, the getCharacteristics() method of a services object will check a cached list of characteristics, but the same method for a peripheral object will not. It would be more convent if I could just query one master list of all characteristics.

IanHarvey, 2015-12-08:
Thanks, these are all good points.

Some things (e.g. the 'iface' parameter) are recently added, or got forgotten, and will need additions to the docs. I'll be starting this soon.

Other things (e.g. 'discoveredAllServices') are considered private to the implementation and won't appear in the docs. I need to do a pass through the code and make sure all private variables and methods have an initial underscore (PEP 8's "weak internal use" convention) in their names.

None of the classes' data attributes (member variables) are documented - in general I don't like people messing with variables directly: it makes it too hard to change the implementation, and too easy to break things. However, there are some things (UUID.binVal, Service.uuid) which are well-defined and might be useful - these can get turned into (read-only) Python properties and included in the docs.

Thanks
Ian

pcborenstein, 2015-12-08:
I'm glad I could help a bit. Let me know when the scan functionality is ready and I'll test it out

IanHarvey, 2016-01-09:
Hi again,

FYI - I've made some updates and uploaded docs (now at v0.9.11) to http://ianharvey.github.io/bluepy-doc/ - this includes scanning functionality. There is some 'Properties' documentation, but note these aren't proper Python properties in the code yet.

Cheers
Ian

closed 2017-01-30 by IanHarvey