IanHarvey/bluepy #92: Unexpected "Service [..] not found" error

hoonkai, 2015-12-19:
Hi

I'm using TI BLE's SimpleBLEPeripheral code which gives a UUID

`0000fff0-0000-1000-8000-00805f9b34fb`

If sensortag.py is modified to become

```def _TI_UUID(val):
    return UUID("%08X-0000-1000-0000-000000000000" % (0x00000000+val))

...

class SimpleGatt(SensorBase):
    svcUUID  = _TI_UUID(0xFFF0)
    dataUUID = _TI_UUID(0xFFF1)
    ctrlUUID = _TI_UUID(0xFFF2)

    def __init__(self, periph):
        SensorBase.__init__(self, perish)```

and enabling it properly in `main()`, it fails to identify the FFF0 service. The UUID already indicates the FFF0 service and `SimpleGatt` already has `FFF0` as its `svcUUID`. What could be causing the failure?

Any help will be appreciated.

IanHarvey, 2016-03-22:
My guess is that you're somehow not running the code you've just edited, as the service which isn't being found is `f000fff0-0451-4000-b000-000000000000`, and not `0000fff0-0000-1000-8000-00805f9b34fb` (as you say in the comment) or `0000FFF0-0000-1000-0000-000000000000` (as the code you posted will generate).

I would start with:

    class SimpleGatt(SensorBase):
        svcUUID  = UUID("0000fff0-0000-1000-8000-00805f9b34fb")
        print( "Looking for service: " + str(svcUUID))
    ....

Then the code will print exactly what it's looking for, or (if it doesn't print anything at all) you're running the wrong `sensortag.py` somehow.

closed 2017-01-30 by IanHarvey

quique123, 2018-05-28:
Hoonkai, did you ever resolve this issue?