IanHarvey/bluepy #93: Online docs update

Shiselster, 2015-12-26:
Hi there @IanHarvey 
First off, thanks for the great work on this project! It's very helpful :+1: 

I wanted to ask of you to update your docs-site ["**bluepy-doc**"](http://ianharvey.github.io/bluepy-doc/) so it will include the latest changes you've made (in both code and documentation) like the new classes Scanner, ScanEntry and DefaultDelegate as well as the other changes you've made between *0.9.0* and *0.9.1* like in the Peripheral class

I know the changes already reflect in the .rst files, they just need to be re-"compiled" again by Sphinx and uploaded to the website :)

Thanks ahead!
Shisel

IanHarvey, 2016-01-09:
I've just built the latest docs - now at version 0.9.11 - and uploaded them to http://ianharvey.github.io/bluepy-doc/. 

This has documentation for the new scanning features.

Cheers
Ian

closed 2016-01-09 by IanHarvey

Shiselster, 2016-01-18:
check: can i re-open a closed issue?

Shiselster, 2016-01-18:
re-opening