IanHarvey/bluepy #94: Scan error

Ivan-Perez, 2015-12-28:
Hi!

I'm having trouble to scan devices, using the new `Scanner` class I've found after updating today the library. I've recompiled the helper, but it didn't help. Other features in the library are working properly (such reading and writing from Bluetooth devices).

Here is my code:
```python
from btle import UUID, Peripheral, Scanner

sc = Scanner()
sc.scan(timeout=5)
```

It gives me the next error:
```
Traceback (most recent call last):
  File "test_scan.py", line 4, in <module>
    sc.scan(timeout=5)
  File ".../bluepy/bluepy/btle.py", line 631, in scan
    self.start()
  File ".../bluepy/bluepy/btle.py", line 572, in start
    self._mgmtCmd("le on")
  File ".../bluepy/bluepy/btle.py", line 240, in _mgmtCmd
    "Failed to execute mgmt cmd '%s'" % (cmd))
btle.BTLEException: Failed to execute mgmt cmd 'le on'
```

Am I doing anything wrong? Thank you!

adminq80, 2016-01-07:
I using Python 3.4 and I received the following trace

    Traceback (most recent call last):
        File "scanner.py", line 17, in <module>
        s.scan(timeout=10)
      File "/src/bluepy/bluepy/btle.py", line 632, in scan
        self.process(timeout)
      File "/src/bluepy/bluepy/btle.py", line 624, in process
        self.delegate.handleDiscovery(dev, (dev.updateCount <= 1), isNewData)
    AttributeError: 'DefaultDelegate' object has no attribute 'handleDiscovery'

When I looked at the file I saw a strange line of code (line 623)

```python
    if self.delegate or True:
        self.delegate.handleDiscovery(dev, (dev.updateCount <= 1), isNewData)

IanHarvey, 2016-01-20:
Regarding the original issue:
- You need to run as root in order to do scanning.
- If you set 'Debugging=True' in btle.py, it will show much more information. If this doesn't reveal an obvious problem, post the trace here and I'll take a look at it.

Thanks
Ian

Ivan-Perez, 2016-01-23:
Thank you for your answer! Executing the script as root fixed the problem.

Maybe a better message in the exception can help to other users facing this problem.

closed 2016-01-23 by Ivan-Perez