IanHarvey/bluepy #96: Bluepy bitbake recipe error

kerimcagri, 2016-01-07:
Hi,
I am trying to write bitbake recipe for bluepy, I downloaded source and compiled on my host machine and it worked fine. However, when I try to build the package for intel-baytrail gw via build environment (windriveridp3) --despite I only call make command on Makefile-- the ld can not find the libc.so.6 and libc_nonshared.a. Libc exists in target sysroot but I dont know why ld is looking for /lib64 directory instead sysroot I provided(or I think I provided).

Error is as follows:

| cannot find /lib64/libc.so.6
| /usr/bin/ld: cannot find /usr/lib64/libc_nonshared.a
| collect2: error: ld returned 1 exit status

Here is a part of my bluepy.bb:

DEPENDS="glib-2.0 dbus"

SRCREV = "${AUTOREV}"
INHIBIT_PACKAGE_DEBUG_SPLIT = "1"
INHIBIT_PACKAGE_STRIP = "1"
SRC_URI = "file://* \
"
S = "${WORKDIR}"
inherit pkgconfig
do_compile() {
	make -C ${S}/bluepy \
	LDFLAGS=" -L${STAGING_DIR_TARGET}/lib64 --sysroot=${STAGING_DIR_TARGET} "
}

IanHarvey, 2016-01-09:
I don't know anything about how bitbake works, but from the information you give it's looking like STAGING_DIR_TARGET is not being set - it's probably looking for *${STAGING_DIR_TARGET}/lib64/libc.so.6*, but because the variable is empty it's coming out as */lib64/libc.so.6*.

IanHarvey, 2016-03-22:
I'd recommend you don't try to use bitbake directly for compiling bluepy-helper; you can either use the Makefile in bluepy itself, or run **setup.py install** (etc) which will do all the setup required. I'm afraid I don't have the time available to debug this any further.

Thanks
Ian

closed 2016-03-22 by IanHarvey