IanHarvey/bluepy #98: Permissions problem

mph070770, 2016-01-13:
Hello. I have the following simple test:

```
from btle import Scanner

sc = Scanner()
sc.scan(timeout=5)
```

and I get the following error:

```
pydev debugger: starting (pid: 14866)
Traceback (most recent call last):
  File "/home/parallels/Projects/liclipse/plugins/org.python.pydev_4.2.0.201507051114/pysrc/pydevd.py", line 2283, in <module>
    globals = debugger.run(setup['file'], None, None)
  File "/home/parallels/Projects/liclipse/plugins/org.python.pydev_4.2.0.201507051114/pysrc/pydevd.py", line 1709, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/home/parallels/Documents/LiClipse Workspace/btle/main.py", line 4, in <module>
    sc.scan(timeout=5)
  File "/usr/local/lib/python2.7/dist-packages/btle.py", line 631, in scan
    self.start()
  File "/usr/local/lib/python2.7/dist-packages/btle.py", line 569, in start
    self._mgmtCmd("le on")
  File "/usr/local/lib/python2.7/dist-packages/btle.py", line 236, in _mgmtCmd
    rsp = self._waitResp('mgmt')
  File "/usr/local/lib/python2.7/dist-packages/btle.py", line 294, in _waitResp
    raise BTLEException(BTLEException.MGMT_ERROR, "Management not available (permissions problem?)")
btle.BTLEException: Management not available (permissions problem?)
```

what am I doing wrong??

mph070770, 2016-01-14:
I've moved the bluepy files into my liclipse workspace and it's working. Not ideal but ok to get me going until I have time to debug why.

closed 2016-01-14 by mph070770

stretchsenselouisgermain, 2017-04-04:
Hey,

I know this subject is close, but I have exactly the same problem as you, the only difference is that I already put the bluepy files into my workspace and  it still doesn't work, can you give me a hand with this on how do you moved the files and make your code work?

abaz21, 2017-07-19:
I'm having this same error. Please let me know if you are able to describe the solution.

brian-lesschwab, 2018-09-19:
@abaz21 @mph070770, have either of you solved this problem? trying to get it to work on a raspberry pi