IanHarvey/bluepy #99: -K switch in sensortag.py causes an error

mph070770, 2016-01-14:
I'm using the sensortag.py example and have it working (I can read temperature with -T switch). However, if I enable keys:

```
python main.py -T -K -t 3 78:A5:04:19:58:D4
```

.. I get the following error. I'm not sure why. Any tips? Thanks:

```
pydev debugger: starting (pid: 2432)
Connecting to 78:A5:04:19:58:D4
Running  /home/parallels/Documents/LiClipse Workspace/btle/bluepy-helper
Sent:  conn 78:A5:04:19:58:D4 public

Got: "rsp=$stat state=$tryconn dst='78:A5:04:19:58:D4 mtu=h0 sec='low\n"
Got: "rsp=$stat state=$conn dst='78:A5:04:19:58:D4 mtu=h0 sec='low\n"
Sent:  wrr 60 0100

Got: 'rsp=$err code=$comerr\n'
Traceback (most recent call last):
  File "/home/parallels/Projects/liclipse/plugins/org.python.pydev_4.2.0.201507051114/pysrc/pydevd.py", line 2283, in <module>
    globals = debugger.run(setup['file'], None, None)
  File "/home/parallels/Projects/liclipse/plugins/org.python.pydev_4.2.0.201507051114/pysrc/pydevd.py", line 1709, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/home/parallels/Documents/LiClipse Workspace/btle/main.py", line 293, in <module>
    main()
  File "/home/parallels/Documents/LiClipse Workspace/btle/main.py", line 263, in main
    tag.keypress.enable()
  File "/home/parallels/Documents/LiClipse Workspace/btle/main.py", line 172, in enable
    self.periph.writeCharacteristic(0x60, struct.pack('<bb', 0x01, 0x00), True)
  File "/home/parallels/Documents/LiClipse Workspace/btle/btle.py", line 459, in writeCharacteristic
    return self._getResp('wr')
  File "/home/parallels/Documents/LiClipse Workspace/btle/btle.py", line 334, in _getResp
    resp = self._waitResp(wantType + ['ntfy', 'ind'], timeout)
  File "/home/parallels/Documents/LiClipse Workspace/btle/btle.py", line 296, in _waitResp
    raise BTLEException(BTLEException.COMM_ERROR, "Error from Bluetooth stack (%s)" % errcode)
btle.BTLEException: Error from Bluetooth stack (comerr)
```

mph070770, 2016-01-14:
It looks like the keypress handle should be 0x6c not 0x60 as shown in sensortag,py below. I've taken this from the TI reference (http://processors.wiki.ti.com/images/a/a8/BLE_SensorTag_GATT_Server.pdf):

```
class KeypressSensor(SensorBase):
    svcUUID = UUID(0xFFE0)
    dataUUID = UUID(0xFFE1)

    def __init__(self, periph):
        SensorBase.__init__(self, periph)
 
    def enable(self):
        self.periph.writeCharacteristic(0x60, struct.pack('<bb', 0x01, 0x00), True)

    def disable(self):
        self.periph.writeCharacteristic(0x60, struct.pack('<bb', 0x00, 0x00), True)
```

IanHarvey, 2016-01-20:
The keypress notifications were working for me on v1.4 firmware, and stopped working in the way you described when I upgraded it to v1.5 (using the iOS Sensortag app). It seems the handle changed between firmware versions.

Commit 04bcf93 fixes this for both v1.4 and v1.5, by basing the 'control' characteristic handle on the 'data' handle (it's 2 higher in both cases). Unlike the other sensors there's no way to discover the control characteristic directly by searching for its UUID, so there's no firm guarantee this will be safe for future firmware versions.

Cheers
Ian

closed 2016-01-20 by IanHarvey