IanHarvey/bluepy #101: intel edison

aafeliz, 2016-01-19:
any idea if it will work for yocto 2.1 on the intel edison?

IanHarvey, 2016-01-27:
There's an Intel document at http://download.intel.com/support/edison/sb/edisonbluetooth_331704004.pdf which suggests that the regular BlueZ 5.x tools will work. The bluepy code is based on BlueZ 5.x, so it ought to work fine. 

It looks like Edison is infested by systemd, so you might find there's some system setup you need to do before anything will work, though...

IanHarvey, 2017-09-25:
Sadly, time was against us, and I never got an Edison to play with before they were discontinued by Intel. It's therefore unlikely that I'll be able to help with any Edison or Yocto-specific issues.

Cheers
Ian

closed 2017-09-25 by IanHarvey