IanHarvey/bluepy #102: Issue with BLE pairing

antonkluev, 2016-01-21:
Guys,

i try to connect RPI via USB BLE Dongle to BLE Modul on my Arduino... ive tryed all the tutorials in internet, and googled all the issues that i received, but no success! Even i purchased another BLE USB Dongle in Case that the another one is defect or something like this... 

Mainly ive used this tutorial http://www.jaredwolff.com/blog/get-started-with-bluetooth-low-energy/
And i receive this error: 

antonkluev.pi ~ $ sudo hcitool lescan
LE Scan ...
00:15:83:00:41:3A (unknown)
00:15:83:00:41:3A (unknown)
antonkluev.pi ~ $ sudo gatttool -b 00:15:83:00:41:3A -I -t random
[00:15:83:00:41:3A][LE]> connect
Attempting to connect to 00:15:83:00:41:3A
Error: connect: Connection refused (111)
[00:15:83:00:41:3A][LE]>

if i erase "-t random" than the error is the same... i have an assumption that the "-t" flag not working... I am new in this technology, therefor, realy sorry for my not knowing, i need realy help...

0xFACE, 2016-01-22:
Try
<code>btmgmt le on</code>
The btmgmt command is part of bluez package, but is not installed by default.

antonkluev, 2016-01-22:
thank you 0xFACE :) i know what was wrong! My kernel version was 4.+ and this was an issue. Somehow this kernel version is not compatible with Bluez or BLE.. don't realy know )) 
Ive installed wheezy Raspbian with commone 3.18.11 and "it works like a charm" XD

Did you try to run BLE on 4.+ kernel?

0xFACE, 2016-01-23:
Yes ;-)
<code>
Linux raspberrypi 4.1.15+ #831 Tue Jan 19 18:36:09 GMT 2016 armv6l GNU/Linux
</code>

antonkluev, 2016-01-23:
omg! 0xFACE! i wannt to kiss you, man! Jesus! It works! You don*t have an idea, how long ive sittet with this issue... and now you enter with you simple solution! Crazy :)

once again - thank you!

0xFACE, 2016-01-23:
Before I found the solution just by playing with bluez commands I spend few weeks googling without success.
I'm glad to see it works also for you ;-)

closed 2017-09-30 by IanHarvey

957919019, 2019-03-01:
you can try Use 'SystemClock.sleep(500)'  to avoid operation to fast

saviourcode, 2020-06-12:
> Try
> `btmgmt le on`
> The btmgmt command is part of bluez package, but is not installed by default.

Thanks alot, it worked!

knito, 2021-01-06:
> Try
> `btmgmt le on`
> The btmgmt command is part of bluez package, but is not installed by default.

This is the solution to see my BLE device in the list! Many thanks!!!!

Jackfritt, 2021-03-31:
Had the same Problem and after using this with a BT 4.0 noname Dongle and PC hardware with latest bluez5.54 it worked like a charm. One day searching and trying hard! Thx.