IanHarvey/bluepy #104: Support for older bluez versions?

Frostlock, 2016-02-02:
Hi all,
I'm running Lubuntu 14.0.4 which has a bluez 4.x from the repositories.
My tests with bluepy so far appear to be OK with this version of bluez but I'm wondering if this could cause difficulties for me? Should I upgrade to a higher version of bluez?

I do have some disconnects I can't explain every now and then and I wonder if it might be linked to my bluez version... It could also be me sending faulty commands to my parrot drone causing it to disconnect or crash :-)

IanHarvey, 2016-04-14:
In general, `bluepy` doesn't really care which version of bluez is used by the system - it has its own copy of all the bluez sources which it uses. These talk directly to the kernel's socket interface, and not for instance the D-Bus API, which will change between bluez versions.

Frostlock, 2016-04-15:
Great, thanks for clarifying!

closed 2016-04-15 by Frostlock