IanHarvey/bluepy #105: issues when try to connect

dammianhr, 2016-02-09:
i have this error in raspberry pi b+ with jessie:

pi@raspberrypi:~/bluepy/bluepy $ python btle.py B0:B4:48:BC:F3:05
Connecting to: B0:B4:48:BC:F3:05, address type: public
Traceback (most recent call last):
  File "btle.py", line 683, in <module>
    conn = Peripheral(devAddr, addrType)
  File "btle.py", line 318, in __init__
    self.connect(deviceAddr, addrType, iface)
  File "btle.py", line 367, in connect
    "Failed to connect to peripheral %s, addr type: %s" % (addr, addrType))
__main__.BTLEException: Failed to connect to peripheral B0:B4:48:BC:F3:05, addr type: public

pi@raspberrypi:~/bluepy/bluepy $ python sensortag.py B0:B4:48:BC:F3:05 -n 5 -t 0.5 -T
Connecting to B0:B4:48:BC:F3:05
Traceback (most recent call last):
  File "sensortag.py", line 298, in <module>
    main()
  File "sensortag.py", line 252, in main
    tag = SensorTag(arg.host)
  File "sensortag.py", line 184, in __init__
    Peripheral.__init__(self,addr)
  File "/home/pi/bluepy/bluepy/btle.py", line 318, in __init__
    self.connect(deviceAddr, addrType, iface)
  File "/home/pi/bluepy/bluepy/btle.py", line 367, in connect
    "Failed to connect to peripheral %s, addr type: %s" % (addr, addrType))
btle.BTLEException: Failed to connect to peripheral B0:B4:48:BC:F3:05, addr type: public
thanx for share!

namoaton, 2016-03-02:
@dammianhr Hello. Try to  set random  address type. Hope this will be helpfull for you

jardous, 2016-03-06:
same problem here. Does not help to set the address type to random:

__main__.BTLEException: Failed to connect to peripheral CD:24:BA:5A:0A:40, addr type: random

0xFACE, 2016-03-06:
Can you connect with gatttool ?
What kernel version?
` uname -r`

jardous, 2016-03-06:
yes - I can connect with gatttool (using the -t random)

```
$ uname -r
4.1.18-v7+
```

namoaton, 2016-03-06:
try to use PUBLIC  address type

jardous, 2016-03-06:
that is default. BUT - it seems it started working after the peripheral restart. Thanks!

0xFACE, 2016-03-06:
For newer kernels I use 
`btmgmt le on`
before attempt a LE connection
But if gatttool works for you, that's not a case.

IanHarvey, 2016-03-22:
I'll close this, looks like it's now fixed.

closed 2016-03-22 by IanHarvey

vaibhavkambli, 2018-02-12:
```
 list = ['24:71:89:E8:F0:84', '24:71:89:E9:44:06']
    #list = ['24:71:89:E8:F0:84']
    i=0
     
    while True:
        print("Debug: Level- While True Loop\n")
        tag = SensorTag(list[i])
        print("tag value ", tag)
        print('Sensor ID: ', list[i])

        tag.IRtemperature.enable()
        tag.humidity.enable()
        tag.barometer.enable()
        tag.accelerometer.enable()
        tag.magnetometer.enable()
        tag.gyroscope.enable()
        tag.battery.enable()
        ##    tag.keypress.enable()
        ##    tag.setDelegate(KeypressDelegate())
        tag.lightmeter.enable()

        time.sleep(1.0)
        print('Temp: ', tag.IRtemperature.read())
        print("Humidity: ", tag.humidity.read())
        print("Barometer: ", tag.barometer.read())
        print("Accelerometer: ", tag.accelerometer.read())
        print("Magnetometer: ", tag.magnetometer.read())
        print("Gyroscope: ", tag.gyroscope.read())
        print("Light: ", tag.lightmeter.read())
        print("Battery: ", tag.battery.read())

        tag.disconnect()
            
        del tag
        time.sleep(2)
        i+=1
        if i == 2: #add sensors in  last increment this count
            i=0
```

This is small code i am using to access TI 2650 sensor nodes.
If node is not activated i.e. they are in sleep mode they will through error and exit the program,
can anyone suggest any method so that i can go and check if sensor in sleep or not , if in sleep do not try to access it. 

 I am getting below error, in long run test of 1hr to 2hr, even same error when node is in sleep and i tried to access it.

[image]

subvertallchris, 2019-02-04:
Old old post but switching address type to random did the trick for me.

chewkahmeng, 2020-03-13:
@subvertallchris can you show me how you changed the address type to random? Thanks!

bwomsm1, 2023-03-15:
Hi, did anyone find any solution for this issue? 
I am dealing with that issue for 3 days already, trying to run my code from Raspberry PI 4.

thanks