IanHarvey/bluepy #107: Encountered issuses with sensortag CC2650

Jason-Gew, 2016-02-16:
Hi Ian,

I am trying to use sensortag.py to enable services and read corresponding data from TI SensorTag CC2650. When I tried to read barometer, it showed some issues with service enable. 
[image]

Besides, I also encountered a problem with reading the accelerometer data. The result showed service f000aa10-0451-4000-b000-000000000000 not found (from btle.ETLEException).

However, reading temperature, humidity and key-press are functioning. So I am wondering whether there's some where I operated wrong  or the handle needs to be updated?

Is there any ideas about that? :)

IanHarvey, 2016-02-19:
Hi Jason - can you try doing a git pull (which should update your _sensortag.py_) and trying again?
It looks like I hadn't pushed the CC2650 support properly; it was present in tag v/0.9.12 but not on master - sorry about that.

Cheers
Ian

Jason-Gew, 2016-02-19:
Thank you Ian, I update the sensortag.py and it supports the CC2650 now. You did a really excellent work!

IanHarvey, 2016-02-20:
Thanks for letting me know - have fun!
Ian

closed 2016-02-20 by IanHarvey