IanHarvey/bluepy #109: Can bluepy act as different roles?

namoaton, 2016-03-01:
Can bluepy act as different roles?

IanHarvey, 2016-03-03:
No, there isn't any support for acting as a Peripheral at the minute.
A GATT server will be quite a different 'shape' to the current API. I'm debating whether to keep it in the existing framework (with `bluepy-helper`), or whether it should be a complete re-implementation based on `asyncio` or similar.

namoaton, 2016-03-03:
It would be greatfull  if bluepy can act as peripherial. I can help to test such feature

smdjeff, 2016-04-10:
BlueZ is just about impossible to get working as a peripheral. I think something was deprecated that causes "register GATT service with handle" when the master tries to read services. It sure would be nice to have BluePy and less BlueZ dependent alternative for the new Raspberry Pi 3 which has BT and LE out of the box!

IanHarvey, 2016-04-14:
I've spent quite a lot of effort trying to get BlueZ peripheral code to do anything interesting, without success.

I'm beginning to think that writing a whole Python stack from scratch is the way to go. I note there's a Javascript module (https://github.com/sandeepmistry/bleno) which does just this, and doesn't use anything like BlueZ.

wernerterreb, 2016-04-14:
I'm also at the moment starting to doubt suitability of Bluez for the stuff I'm trying to do.  Basically I need the pair securely with a peripheral device that requires exchange of Identity Resolving Keys (IRK).  But try as I like, I cannot seem to make it work on Bluez.   But it was relatively trivial to do on Android.

IanHarvey, 2017-11-13:
Closing this issue. `bluepy` is not going to support the Peripheral role in future versions.

I'd recommend interested parties look at the `bleson` project (current first drafts at https://github.com/TheCellule/python-bleson/blob/wk_draft_experiment/README.rst) for Python beacon and peripherals support.

Thanks
Ian

closed 2017-11-13 by IanHarvey