IanHarvey/bluepy #110: Notification with multiple connections

namoaton, 2016-03-02:
Notification with multiple connections. How can we easily distinguish  without cHandle, but with  for example addr?

IanHarvey, 2016-03-03:
The easiest thing is just to add a reference to the `Peripheral` in your delegate class, e.g.

    class MyDelegate(DefaultDelegate):
        def __init__(self, peripheral):
            DefaultDelegate.__init__(self)
            self.peripheral = peripheral
    
    ...
    dev = Peripheral(addr)
    dev.setDelegate( MyDelegate(dev) )

You will then have a `self.peripheral` object available in your delegate's `handleNotification`.

namoaton, 2016-03-03:
I will try it today and then write here about my experience

namoaton, 2016-03-03:
when I print peripheral.addr  it is returned None

namoaton, 2016-03-03:
I proposed to add get_addr() function to Peripheral class

closed 2016-03-07 by namoaton