IanHarvey/bluepy #113: ATT_MTU > 20 bytes

MichaelSteurer, 2016-03-10:
Hello Ian, hello everyone,
I have a problem reading and writing characteristics where the packet size is > 20 bytes. Both requests fail on my Raspberry PI although the BLE hardware supports it (verified with an Apple mobile device). According to the specification, BLE supports up to 512 bytes for reading and writing.

Searching the Internet for my problem brought me to the MTU size which seems to be too small to carry more than 20 bytes. Still, I couldn't figure out how to set it or configure bluez correctly.

What am I missing to configure bluepy or bluez to transfer more than 20 bytes?

Cheers
  Michael

brain4d, 2016-08-26:
Maybe you want to use setMTU to set the size of btle peripheral
Easy,

p = btle.Peripheral('00:00:00:00:00:00',btle.ADDR_TYPE_PUBLIC)
p.setMTU(40) #set the size of mtu as 40

wofcardoso, 2019-04-10:
I am facing the same problem, working with a Scale I send 8 bytes the scale responds with  a 34 bytes, did anyone find a solution to read that many bytes?

MichaelSteurer, 2019-04-10:
Back then we haven't been successful either and used the Apple device as a workaround. Since then the Raspberry 3 Model B+ was released that supports BLE Specification 4.2 - maybe you can find someone with such a device.
@brain4d sorry, I missed your posting three years ago; thanks for getting back. :-)

alberk8, 2019-06-10:
Have you managed to get that to work?  Thank you.

> I am facing the same problem, working with a Scale I send 8 bytes the scale responds with a 34 bytes, did anyone find a solution to read that many bytes?

wofcardoso, 2019-06-10:
No I did not, I was integrating with a TAIDOC scale and it return more
than  32 bytes, problem with that is that the bluepy stack only read up to
16 bytes (me thinks) even if I keep reading the output in a loop it keeps
return the same lenght. Maybe I will use another bluetooth stack that allow
me to read more than 8 bytes.