IanHarvey/bluepy #115: Failed when changing interface to use

cijay, 2016-03-18:
When I try to pass by argument an other BLE interface : `Peripheral("11:22:33:ab:cd:ed", iface=0)`
It still use the default BLE interface and not hci0 like I ask.

I tried to investigate a little more and I got that from `bluepy-helper` in debug mode

```
$ sudo ./bluepy-helper 0
# main() :bluepy-helper.c built at 17:10:04 on Mar 18 2016
# mgmt_setup() :Setting up mgmt on hci0
# main() :Starting loop
# mgmt_debug() :mgmt: [0x0000] command 0x0001
# mgmt_debug() :mgmt: > 02 00 00 00 03 00 01 00 11                       .........       
# mgmt_debug() :mgmt: [0x0000] command 0x01 status: 0x11
# read_version_complete() :Failed to read version information: Invalid Index (0x11)
```

Any idea ?

IanHarvey, 2016-03-22:
I don't understand what you're asking for here. When you use `Peripheral( ..., iface=0)` you are requesting to use the management interface on `hci0`. This is also the default interface. `bluepy-helper` appears to be using `hci0` exactly as requested.

The 'Invalid Index' error is a bug, but it's not actually a harmful one. The code concerned doesn't do anything more than report the controller version information in the debug output. I'll check in a fix, but it won't change any other behaviour.

cijay, 2016-03-22:
Maybe a screenshot will be more understandable

[image]

On the right, init bluepy-helper to use the hci0 interface and connect to a device
On the left, dump of hci0 and hci1.

Only hci1 seems to be used, and not hci0 like requested

(hci0/hci1 support BLE and I am using Ubuntu 14.04)

IanHarvey, 2016-03-22:
Aha, I think I might have spotted a problem.
If you type `conn cd:69:72:c2:0e:19 public hci0` into `bluepy-helper`, does it do the right thing? The Python code will be supplying `0` instead of `hci0` as the last argument here.

cijay, 2016-03-22:
yep, works great with `hci0` instead of `0`

Thanks

IanHarvey, 2016-03-23:
I'll reopen this, because it'll need a fix to `btle.py` to pass the argument in correctly when it starts `bluepy-helper`. I'll close it again when the fix is pushed.

commit c7999082 (2016-03-26): Fix #115: pass 'hciX' as arg to bluepy-helper; also, add --hci option to blescan for testing
closed 2016-03-26 by IanHarvey