IanHarvey/bluepy #117: Is there a way to configure the connection interval, slave latency and supervisor timeout? (hcitool lecup)

shailurie, 2016-03-21:


gonzob, 2016-05-27:
Hi. I'm not sure what the originator of this issue meant, but I'm trying to find a way of setting the above parameters from within BluePy.

I have successfully connected to a Yamaha BLE MIDI device (UDBT01)  from a Raspberry Pi using BluePy. The UDBT01 is plugged in to my AKAI EWI-USB wind controller and it talks via Bluetooth to the synth running in my RPi. It works, but unless I re-configure the BLE connection parameters after the connection is made, the latency is terrible. 

I run:
sudo hcitool lecup --handle $HANDLE --min 6 --max 7 --latency 0 --timeout 500
from within a bash shell ($HANDLE is the current connection handle), and the latency becomes acceptable.

So my question is: can the parameters be set in BluePy as part of the connection setup?

Ben

cjntaylor, 2017-01-25:
Is there a way to get the connection handle integer from a peripheral, in the meantime? All you'd need is a way to get this identifier and you could run this using subprocess.Popen as a stopgap...

IanHarvey, 2017-02-02:
The short answer is, I think, no. The connection handle (the thing you get out of the _Connection Complete_ event) doesn't ever come out of the internals of the AF_BLUETOOTH socket, as far as I can tell. 

@gonzob - you're getting the connection handle by looking at hcidump output, or something??

It's looking increasingly like I'm going to give up on BlueZ and write a whole stack just in Python, talking to a raw HCI socket.

gonzob, 2017-02-02:
Hi Ian.
I use "hcitool con" to find the handle. I have it in a loop which checks every 5 secs that the last handle is still in place. If there's a new one (i.e. the connection failed and was re-started) it re-initialises the settings with hcitool as above.
My project site is here:  http://projects.gonzos.net

Ben

OrenLederman, 2017-02-03:
We had a similar issue in our project, and we ended up changing these files to control the connection intervals :

echo 16 > /sys/kernel/debug/bluetooth/hci0/conn_min_interval
echo 17 > /sys/kernel/debug/bluetooth/hci0/conn_max_interval

you only need to set the value once, which is easier than calling hcitool

(Credits go to Andrew Bartow)

gonzob, 2017-02-03:
That looks great! However, on the two systems I have checked (ubuntu and raspbian) those files seem to be re-created each time the bluetooth connection comes up. Won't your values be overwritten?

Ben

OrenLederman, 2017-02-03:
Actually, I never noticed that. My code sets these values every time anyway :)

ldk16, 2017-05-16:
According to my experience, the parameters set through such method would be reset when the computer restarts.

yupeng-zglue, 2017-09-13:
echo 16 > /sys/kernel/debug/bluetooth/hci0/conn_min_interval
echo 17 > /sys/kernel/debug/bluetooth/hci0/conn_max_interval

My BLE master was sending a Connection Request with the correct interval values, after I had changed the above 2 files. However, it always accepts the Connection Parameters suggested by the slave.

Do you all know how to reject Connection Parameters Update Requests from the slave?

pstanton, 2017-11-29:
i think i was observing the same as @yupeng-zglue so i worked out how to detect a new connection in python and then run the hcitool method from python. it's working well.

yupeng-zglue, 2017-11-30:
@pstanton 
"worked out how to detect a new connection" 
By this you meant detecting when the BLE slave tries to update the connection paramters?

pstanton, 2017-11-30:
yes, specifically "PropertiesChanged" > "org.bluez.Device1" > "ServicesResolved"