IanHarvey/bluepy #120: Battery Level characteristic handle as binary instead of hex

souellet-ms, 2016-03-23:
I was trying to read the Battery Level characteristics of the Bluetooth Battery Service and was reading 'd' when I should have read 0x64 for 100%. While digging into bluepy, I found that when reading the value handle of this characteristics (Sent:  rd 1A), bluepy-helper returned the correct value (Got: 'rsp=$rd d=b64\n'), but though the value was binary 'b' when it is actually and hexadecimal value.

From this point, I am not entirely sure if the problem originated in bluepy-helper or elsewhere (bluez maybe).

Any idea ?

louiscaron, 2016-03-23:
Hello souellet
All you are getting in read response is correct.  You are receiving the string "d" which corresponds to a 1 byte string with char d.  And char d in the ascii table corresponds to 0x64 = 100.
This is normal because when you read an attribute, you do not always know what the content of the attribute is, and it is up to the user to perform the conversion, therefore, the read function always returns a string of bytes.

When you read the battery level for example, you know that you are going to retrieve a single byte so you can extract the value by unpacing it.
rv = batt_char.read()batt_level = struct.unpack("<B", rv)[0]
for example
Louis

souellet-ms, 2016-03-23:
That was it, thanks.

closed 2016-03-23 by souellet-ms