IanHarvey/bluepy #123: Scan without timeout

JPMJPM, 2016-04-02:
Hi, this is not an issue, it is a question. It is possible to scan without timeout ?. 

Regards.

IanHarvey, 2016-04-14:
Yes, basically if you have a `Scanner` object, set a delegate for it, and call its `.start()` method, then you can just call `.process()` over and over. The delegate will be called when any scan results are received.

(I've just noticed the docs at http://ianharvey.github.io/bluepy-doc/scanner.html aren't up-to-date. Sorry about that, I'll leave this issue open until it's fixed).

JPMJPM, 2016-04-15:
Hi, thank you for the answer.

IanHarvey, 2016-04-24:
I've just updated the Scanner docs, so it should have the fixed example code in it. 

To scan without timeout, you can change the example code to look like this:
```python
scanner = Scanner().withDelegate(ScanDelegate())
scanner.start()
while True:
    print "Still running..."
    scanner.process()
```

As this code never completes, you will of course have to do useful things in the ```handleDiscovery()``` method of your delegate.

closed 2016-04-24 by IanHarvey

JPMJPM, 2016-04-24:
Thanks, this is exactly what I need.

Regards.

mrigankmishra, 2018-05-29:
Hi,

We are attempting to connect to a BLE Device via Pi3 through Bluepy and are getting the error - _Select Timeout_. Below is an excerpt of the logs being generated:

Got: 'rsp=$mgmt code=$success\n'
Got: "rsp=$stat state=$scan dst='(null) mtu=h0 sec='low\n"
Got: 'rsp=$scan addr=b0A923B0824C9 type=h2 rssi=h48 flag=hFFFFFFFC
d=b1EFF06000109200258EB2D68BEFB280E7A4BCBD23BA8FF1CB15794916FE3F4\n'
Got: 'rsp=$scan addr=b3D7E9BE51007 type=h2 rssi=h51 flag=hFFFFFFFC
d=b1EFF0600010920028944494046B576FE5DC5DE79018C49C4C936234C25F7A4\n'
Got: 'rsp=$scan addr=b33F00A3762B5 type=h2 rssi=h53 flag=hFFFFFFFC
d=b1EFF060001092002F604564658AC54407224ACF6B92D7F839AC690D3A88066\n'
Got: 'rsp=$scan addr=bFDEDB984F165 type=h2 rssi=h25 flag=h0
d=b050852545F54020106\n'
DATA READ:  ffcb113b01031118257064040a0b1485000001
Device Serial No: 11182570, battery Percentage: 100%, Temperature:
25.71°C, Humidity: 52.53%, Alarm Type: 01
Inserted Flag:  1324
Got: 'rsp=$scan addr=b3FF7FE983DF4 type=h2 rssi=h5C flag=hFFFFFFFC
d=b1EFF06000109200079B864563C3322B2EA9D6B83F754CA244EBAB6DA380B9C\n'
Got: 'rsp=$scan addr=bF07D6D3F6AF2 type=h2 rssi=h39 flag=h0
d=b050852545F540201061416FFCB113B010311182567640409FE148D000001\n'
DATA READ:  ffcb113b010311182567640409fe148d000001
Device Serial No: 11182567, battery Percentage: 100%, Temperature:
25.58°C, Humidity: 52.61%, Alarm Type: 01
Inserted Flag:  1325
Got: 'rsp=$scan addr=b3AEFC164387D type=h2 rssi=h65 flag=hFFFFFFFC
d=b1EFF06000109200293250EDA9153FF9197D37A44145745852269CAA09D76CC\n'
Acknowledge Received:  1324
Got: 'rsp=$scan addr=bFDEDB984F165 type=h2 rssi=h41 flag=hFFFFFFFC
d=b1416FFCB113B01031118257064040A0A1482000001\n'
DATA READ:  ffcb113b01031118257064040a0a1482000001
Device Serial No: 11182570, battery Percentage: 100%, Temperature:
25.7°C, Humidity: 52.5%, Alarm Type: 01
Inserted Flag:  1326
Select timeout
Got: 'rsp=$scan addr=b65F68EDE55A5 type=h2 rssi=h50 flag=h0
d=b02011A0AFF4C0010050B10EE6066\n'
Acknowledge Received:  1325
Acknowledge Received:  1326
Got: 'rsp=$scan addr=b2918AD8EF797 type=h2 rssi=h5C flag=hFFFFFFFC
d=b1EFF0600010920029489A83FAD179530ECE71DD01E06E6401CFC17100A2A34\n'
Got: "rsp=$stat state=$disc mtu=h0 sec='low\n"
Sent:  scan

Got: 'rsp=$mgmt code=$success\n'
Got: "rsp=$stat state=$scan dst='(null) mtu=h0 sec='low\n"
Got: 'rsp=$scan addr=b33F00A3762B5 type=h2 rssi=h57 flag=hFFFFFFFC
d=b1EFF060001092002F604564658AC54407224ACF6B92D7F839AC690D3A88066\n'
Got: 'rsp=$scan addr=b65F68EDE55A5 type=h2 rssi=h5B flag=h0
d=b02011A0AFF4C0010050B10EE6066\n'
Got: 'rsp=$scan addr=b0A923B0824C9 type=h2 rssi=h49 flag=hFFFFFFFC
d=b1EFF06000109200258EB2D68BEFB280E7A4BCBD23BA8FF1CB15794916FE3F4\n'
Got: 'rsp=$scan addr=b3D7E9BE51007 type=h2 rssi=h50 flag=hFFFFFFFC
d=b1EFF0600010920028944494046B576FE5DC5DE79018C49C4C936234C25F7A4\n'
Got: 'rsp=$scan addr=bFDEDB984F165 type=h2 rssi=h36 flag=h0
d=b050852545F540201061416FFCB113B01031118257064040A0A1482000001\n'
DATA READ:  ffcb113b01031118257064040a0a1482000001
Device Serial No: 11182570, battery Percentage: 100%, Temperature:
25.7°C, Humidity: 52.5%, Alarm Type: 01
Inserted Flag:  1327
Acknowledge Received:  1327
Got: 'rsp=$scan addr=b3FF7FE983DF4 type=h2 rssi=h60 flag=hFFFFFFFC
d=b1EFF06000109200079B864563C3322B2EA9D6B83F754CA244EBAB6DA380B9C\n'
Got: 'rsp=$scan addr=bF07D6D3F6AF2 type=h2 rssi=h38 flag=h0
d=b050852545F540201061416FFCB113B010311182567640409FD148D000001\n'
DATA READ:  ffcb113b010311182567640409fd148d000001
Device Serial No: 11182567, battery Percentage: 100%, Temperature:
25.57°C, Humidity: 52.61%, Alarm Type: 01
Inserted Flag:  1328
Select timeout
Acknowledge Received:  1328
Got: 'rsp=$scan addr=b2918AD8EF797 type=h2 rssi=h4C flag=hFFFFFFFC
d=b1EFF0600010920029489A83FAD179530ECE71DD01E06E6401CFC17100A2A34\n'
Got: 'rsp=$scan addr=b3AEFC164387D type=h2 rssi=h62 flag=hFFFFFFFC
d=b1EFF06000109200293250EDA9153FF9197D37A44145745852269CAA09D76CC\n'
Got: 'rsp=$scan addr=bAC233F3003E2 type=h1 rssi=h5E flag=h0
d=b0201061AFF4C000215FFD15047EF4340BD9D5F18D758E79B7C000501D0C5\n'
Select timeout
Select timeout
Select timeout
Select timeout
Select timeout
Select timeout
Select timeout
Select timeout
Select timeout

We are calling .process() continuously without passing any timeout value explicitly. Still the scanning gets stuck with the Select Timeout error. Kindly assist.