IanHarvey/bluepy #127: bluepy-helper 1.0.4 wrong executable format from pip

souellet-ms, 2016-04-22:
I just installed bluepy 1.0.4 via the python3-pip on a fresh Ubuntu 16.04 install while using Python 3.5.

On execution, I received an OSError: Exec format error when executing bluepy-helper.

When checking the executable (using: file bluepy-helper) I got:
bluepy-helper: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=bb101bae1661c9f46a35e08eb4d10f43fffef2ad, not stripped

Is it possible that the executable is built for ARM (probably built for raspberry pi) and not usage on a x64 machine ?

IanHarvey, 2016-04-24:
I've now released 1.0.5 to PyPI, which has a fix for the problem. It works for me on x86 Linux (LinuxMint 17.2). Note you'll need to run:
```
sudo apt-get install build-essential libglib2.0-dev
```
or similar beforehand, otherwise there will be compilation errors.

souellet-ms, 2016-04-25:
Your fix solved it.

FYI, bluepy does work well under Ubuntu 16.04 x64 and built-in bluez 5.37.

Keep up the good work !

closed 2016-04-25 by souellet-ms