IanHarvey/bluepy #129: the maximum of peripheral = 3?

orz811017, 2016-05-07:
I wanna to test the maximum of peripheral with one central

but when I tested on my dongle, I got an error.

(dongle has connected with 3 peripherals, but failed in connecting with 4th)

thanks!!

mbifulco, 2016-08-04:
Would love to hear more about this, @IanHarvey - I'm looking at using bluepy for a project that would require connections to many BLE devices at a time... upwards of 50.  Is there a limit on connections baked into your code?

swist, 2016-08-04:
A bit of an unrelated note, but you might find it helpful - when I was working with a similar BLE library for node.js it turned out that despite no hard limit in the code, we were struggling to get more than 10 devices maintain connection at once. There were multiple factors that influenced it, ranging from the type of USB BLE adapter we used, to how we wrote the code that consumed the library.

mbifulco, 2016-08-04:
@swist that's definitely interesting - tell us more, if you can!

I've got an antenna that is _supposed_ to not have a cap on connections, from (http://aircable.co), but it has a CSR radio, and thus seems to suffer the same limit in practice.

orz811017, 2016-08-06:
I think it is caused by dongle's manufacturers,they have their own ble stack

Originally I used adapter produced by CSR, and it just only connected 3 peripheral at the same time.
I changed the new adpater produced by BroadCom , and now I can connect with more than 8 slaves at the same time(use one dongle).

swist, 2016-08-10:
What OS are you using? We had better results with the linux stak, best results with OS X as it uses the Apple one. 

Good place to ask is: https://github.com/sandeepmistry/noble

People from my previous company use that library that depends heavily on how many slaves you can connect

orz811017, 2016-08-16:
Linux Mint 

thanks ^^

IanHarvey, 2017-08-29:
For clarity - there are no limits in _bluepy_ itself about the maximum number of connected peripherals. Each `Peripheral` has its own instance of `bluepy-helper` which talks to a kernel  socket.

The Bluetooth adapter will have some built-in maximum  (see e.g   https://github.com/sandeepmistry/noble#maximum-simultaneous-connections) but it should be possible to mitigate this by using multiple adapters.

Thanks
Ian

closed 2017-08-29 by IanHarvey