IanHarvey/bluepy #131: Period / Refresh Rate

mritchie712, 2016-05-14:
I'm using bluepy with the TI Sensortag (CC2650) and I'm able to get it to connect and print out data, but the refresh rate is much lower then what I'm seeing on my iOS app. I've reduced the "Movement Period" ( UUID AA83) to 100ms (0x0A), which dramatically increased the number of readings I was getting on the iOS app, but not thru bluepy. I'm running bluepy on a raspberry pi and it doesn't appear the issue is with the pi, is there some config in bluepy that I can change to increase the refresh rate?

mritchie712, 2016-05-14:
I'm an idiot `python sensortag.py xx:xx:xx:xx:xx:xx -n 5 -t 0.5 -T`... `-t` sets the time between polling.

closed 2016-05-14 by mritchie712

ahasan016, 2018-10-08:
hello,
Could you please help me to set the refresh rate inside the get_reading() or enable_sensors() function?