IanHarvey/bluepy #132: Scanner respond slowly while using Raspberry Pi 3 onboard bluetooth

kalacool, 2016-05-19:
Hi,

I had a beacon which can advertise per 100 ms. When I used a Raspberry Pi 2 with the CSR dongle. The scanner(blescan.py) showed the advertisement packets immediately. But when using the Raspberry Pi 3 onboard bluetooth, the scanner showed the advertisement packets only every 10 seconds.

bridger, 2017-01-31:
Were you able to find a solution? I also get very slow updates when scanning. I was hoping to get updates just from scanning (I just need to detect an event without much data) instead of establishing a connection. The updates need to be continuously coming in for this to work, though.

kalacool, 2017-01-31:
Hi,
I just used another project: https://github.com/switchdoclabs/iBeacon-Scanner-
Hope this helps.

mdxs, 2020-10-21:
I think this issue should be closed, it clearly is hardware related; this is not a `bluepy` issue.