IanHarvey/bluepy #136: Bluepy not finding all characteristics

josephroberts, 2016-06-17:
I am using the [bluepy library][1] to communicate with a bluetooth device but it is not showing me all the characteristics.

When I run the following using gatttool I get:

    [EE:50:F0:F8:3C:FF][LE]> char-desc 0x0019 0xffff
    handle: 0x0019, uuid: 00002800-0000-1000-8000-00805f9b34fb
    handle: 0x001a, uuid: 00002803-0000-1000-8000-00805f9b34fb
    handle: 0x001b, uuid: 00001532-1212-efde-1523-785feabcd123
    handle: 0x001c, uuid: 00002803-0000-1000-8000-00805f9b34fb
    handle: 0x001d, uuid: 00001531-1212-efde-1523-785feabcd123
    handle: 0x001e, uuid: 00002902-0000-1000-8000-00805f9b34fb
    handle: 0x001f, uuid: 00002803-0000-1000-8000-00805f9b34fb
    handle: 0x0020, uuid: 00001534-1212-efde-1523-785feabcd123

But when I run the following using Bluepy I get:

    characteristics = self.peripheral.getCharacteristics(startHnd=0x0019, endHnd=0xFFFF, uuid=None)
    for characteristic in characteristics:
        print("{}, hnd={}, supports {}".format(characteristic, hex(characteristic.handle), characteristic.propertiesToString()))

    Characteristic <00001532-1212-efde-1523-785feabcd123>, hnd=0x1a, supports WRITE NO RESPONSE 
    Characteristic <00001531-1212-efde-1523-785feabcd123>, hnd=0x1c, supports NOTIFY WRITE 
    Characteristic <00001534-1212-efde-1523-785feabcd123>, hnd=0x1f, supports READ 

I need to be able to write '\x01\x00' to characteristic:

    handle: 0x001e, uuid: 00002902-0000-1000-8000-00805f9b34fb

followed by writing '\x01\x04' to characteristic:

    handle: 0x001d, uuid: 00001531-1212-efde-1523-785feabcd123

but I am unable to because Bluepy is not finding the characteristic:
    
    handle: 0x001e, uuid: 00002902-0000-1000-8000-00805f9b34fb

Why is this? 

How can I write to the characteristic when Bluepy does not find it? 

Why do the handles returned by Bluepy not match those returned by gatttool?

Thanks

(stack overflow link http://stackoverflow.com/questions/37880453/bluetooth-bluepy-not-finding-all-characteristics)

  [1]: https://github.com/IanHarvey/bluepy

flyingcircle, 2016-09-15:
You are mistaken as to what those additional characteristics are. The uuids that begin with 000028xx are characteristic "declarations" as seen here: [gatt declaration spec](https://www.bluetooth.com/specifications/gatt/declarations), and the uuids that begin with 000029xx are descriptors as seen here: [gatt descriptor spec](https://www.bluetooth.com/specifications/gatt/descriptors). They are not the characteristics itself, but describe details about one of the characteristics.

folkien, 2019-12-09:
What it means? Is it possible to write '\x01\x00' to 
`handle: 0x001e, uuid: 00002902-0000-1000-8000-00805f9b34fb` or not with bluepy? 

This is a real issue for some devices based on notifications.