IanHarvey/bluepy #137: BluePy bogs down with too many notifications?

jimreich, 2016-06-27:
I'm using a  device which takes 2000 data samples and then sends them one at a time via a very fast batch of notifications. I don't have the ability to modify the device behavior, but when I run gatttool, we're able to get those notifications at a high rate, without slowing down.

However, in Bluepy, the notfications slow down after around 500 of them to about 1 notification per second. Is it possible Bluepy just can't keep up, and some buffer is getting filled? Has anyone seen this before, and is there a cure?
.
My handler inherits from DefaultDelegate, and I've disabled all processing of the incoming data -- at the moment, it just increments and prints a counter (doesn't even send a newline, so no scrolling)

jimreich, 2016-06-27:
Aha! The problem wasn't bluepy, it was IDLE.
I'm not sure how Idle hooks up to python, but as soon as I ran direct from the command line, this problem stopped happening.

So remember, folks, for performance-- run from the command line!

closed 2016-06-27 by jimreich

IanHarvey, 2016-06-28:
I think that IDLE's handling of print output is the culprit here. I've noticed before that very long lines of output make it slow down (for reasons unknown). Anyway, glad you fixed it.

jimreich, 2016-06-28:
Hmm, I was actually just printing a single dot (no newline) for each transaction because I feared just that problem. But still slowed to a crawl. So I suspect it’s at the interface between stdout and IDLE, or maybe IDLE just has a debug shim of some sort that makes it run python very slowly. But thanks for bluepy— now that I’m through the problems (for now), it’s making my life a lot easier!