IanHarvey/bluepy #141: ble scan error

kitty840120, 2016-07-26:
Sorry for disturbance. Can I ask you an question about bluepy? 

I use raspberry Pi which with a python inside, and use a bluepy programs to scan ble.

It's doesn't have any bug before. Recently I don't know why compiled of the program after scanning the surrounding ble then have errors.

If I re-opened the Raspberry Pi, then can compiled completely, but after few minutes will perform the same error.

The following are the error that showed like this:

Discovered device 20:cd:39:b3:aa:12

Traceback (most recent call last):
 
 File "ble_scan.py", line 21, in <module>
    
devices = scanner.scan(10.0)
  
File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 632, in scan
    
self.process(timeout)
  
File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 608, in process
    
self._mgmtCmd("scan")
  
File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 240, in _mgmtCmd
    
"Failed to execute mgmt cmd '%s'" % (cmd))

bluepy.btle.BTLEException: Failed to execute mgmt cmd 'scan'

**This is my code**

from bluepy.btle import Scanner, DefaultDelegate

import MySQLdb

import datetime

import json

db=MySQLdb.connect("xxx.xxx.xxx.xxx","xxx","xxx","xxx")

cursor = db.cursor()

time = datetime.datetime.now()

date = datetime.date.today()

class ScanDelegate(DefaultDelegate):  
   
   def __init__(self): 
        
        DefaultDelegate.__init__(self)

   
   def handleDiscovery(self, dev, isNewDev, isNewData): 
        
        if isNewDev: 
            
            print "Discovered device", dev.addr
        
        elif isNewData: 
            
            print "Received new data from", dev.addr

scanner = Scanner().withDelegate(ScanDelegate()) 

devices = scanner.scan(10.0)

for dev in devices:
	      
        #if (dev.addr is "20:CD:39:B3:AA:12"):
        
        #print (time.strftime("%Y/%m/%d"))
        
        #print (time.strftime("%H:%M:%S"))

	
        if(dev.getValueText(9) is "Zero"):
            
            print "Device %s (%s), RSSI=%d dB" % (dev.addr, dev.addrType, dev.rssi)   
	   

for (adtype, desc, value) in dev.getScanData(): 
	    
            print "  %s = %s" % (desc, value)
	    
            cursor.execute("INSERT INTO record values (%s,%s,%s,%s,%s,%s,%s,%s)",('b',date,time,'try',dev.addr,'b','b',dev.rssi))

db.commit()

cursor.close()

db.close()

soysuz, 2016-08-05:
I'm not sure if you're doing anything risky, I don't know much about databases and stuff. Instead of trying to solve problems with ble scanning and bluepy, try to make other parts of your code safer. In my case I couldn't even debug my program, the bluetooth interface got stuck and didn't come back before reboot. Try simple and fail-safe coding for a while, maybe someone with experience may inspect your problem with more detail in the next days

mdxs, 2020-10-21:
This issue should be closed; it appears to be a specific application issue and not something related to `bluepy` itself.