IanHarvey/bluepy #150: Failed to execute mgmt cmd 'scanend'

Owlie, 2016-08-28:
Raspberry pi 3 Model B

Using the example code from http://ianharvey.github.io/bluepy-doc/scanner.html#sample-code
```
from bluepy.btle import Scanner, DefaultDelegate

class ScanDelegate(DefaultDelegate):
    def __init__(self):
        DefaultDelegate.__init__(self)

    def handleDiscovery(self, dev, isNewDev, isNewData):
        if isNewDev:
            print "Discovered device", dev.addr
        elif isNewData:
            print "Received new data from", dev.addr

scanner = Scanner().withDelegate(ScanDelegate())
devices = scanner.scan(10.0)

for dev in devices:
    print "Device %s (%s), RSSI=%d dB" % (dev.addr, dev.addrType, dev.rssi)
    for (adtype, desc, value) in dev.getScanData():
        print "  %s = %s" % (desc, value)

```

I get this error.

```
Running  /usr/local/lib/python2.7/dist-packages/bluepy/bluepy-helper
Sent:  le on

Got: 'rsp=$mgmt code=$success\n'
Sent:  scan

Got: 'rsp=$mgmt code=$busy\n'
Sent:  scanend

Got: 'rsp=$mgmt code=$protoerr\n'
Stopping  /usr/local/lib/python2.7/dist-packages/bluepy/bluepy-helper
Traceback (most recent call last):
  File "main.py", line 18, in <module>
    devices = scanner.scan(10.0)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 631, in scan
    self.start()
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 576, in start
    self._mgmtCmd("scanend")
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 240, in _mgmtCmd
    "Failed to execute mgmt cmd '%s'" % (cmd))
bluepy.btle.BTLEException: Failed to execute mgmt cmd 'scanend'
```

ulfwin, 2016-10-09:
I get the same error running "blescan" in the console. I'm running as root

markrages, 2016-11-15:
This happens on one of my Bluetooth dongles ("0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)") but not the other one ("0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0"). 

The error happens when a random address is set, then scanning is shut down after:
```
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Random Address (0x08|0x0005) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x0c
    Error: Command Disallowed

```
There appears to be a race condition between subsequent invocations of "bluepy-helper".

This patch "fixes" it: (By "fixing" I mean it makes it not happen on my particular computer.  I didn't dig down through the layers of abstraction to solve it properly.)

```
diff --git a/bluepy/btle.py b/bluepy/btle.py
index ec950e1..65e6a78 100755
--- a/bluepy/btle.py
+++ b/bluepy/btle.py
@@ -218,6 +218,7 @@ class BluepyHelper:
             self._helper.stdin.write("quit\n")
             self._helper.stdin.flush()
             self._helper.wait()
+            time.sleep(0.1)
             self._helper = None
         if self._stderr is not None:
             self._stderr.close()
```

brianpeiris, 2017-01-15:
I ran into the error on my raspberry pi but it was because I forgot to enable the bluetooth device in the first place ;/

OdinTech3, 2017-05-03:
I tried to run the same code that the OP ran on a raspberry pi 2 B+ model but then I got this error
`Traceback (most recent call last):
  File "ble_scan.py", line 14, in <module>
    devices = scanner.scan(10.0)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 631, in scan
    self.start()
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 568, in start
    self._startHelper(iface=self.iface)
  File "/usr/local/lib/python2.7/dist-packages/bluepy/btle.py", line 210, in _startHelper
    universal_newlines=True)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory`

EinSoldiatGott, 2017-06-18:
@markrages Your delay also works for a frequent BTL Exception Device Disconnected when scanning Frequently.

I understand that it's a workaround but it helped to eliminate the frequent exceptions.

Thanks

okelet, 2017-09-13:
Same here, but the patch doesn't work for me... :( Ubuntu 16.04 x64, bluepy version 1.1.2.

okelet, 2017-09-13:
If this helps... This error has occurred when Bluetooth was disabled on my laptop (I didn't realize it). If I enable BT, I don't get that error...

vidvisionify, 2018-07-31:
Also running into this on Ubuntu Server 18.04

lokeshh, 2019-01-17:
Try `sudo hciconfig hci0 down && sudo hciconfig hci0 up`.

j616, 2019-08-26:
Also finding this issue. It seems to happen particularly often under docker. Worth noting that restarting the interface with hciconfig isn't practical for long running logging applications etc.

johngo7470, 2019-11-12:
Yes, I have this issue, too. Raspian 4.19, onboard bt.  While hci down/up can clear it, like j616 commented, I need it for a persistent logging application (monitors motion sensors).

johngo7470, 2019-11-12:
My Pi4 also had issues - the process hung after running for a little while.  I ended up switching to a USB-based BT devices (BLED112) with bled112_scanner.py.

rakeshscetic, 2019-11-15:
I have same issue and it run for while and give me Error: 

Traceback (most recent call last):
  File "TempHumMonitor.py", line 46, in <module>
    devices = scanner.scan(10.0)  
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 852, in scan
    self.start(passive=passive)
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 797, in start
    self._mgmtCmd(self._cmd()+"end")
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 312, in _mgmtCmd
    raise BTLEManagementError("Failed to execute management command '%s'" % (cmd), rsp)
bluepy.btle.BTLEManagementError: Failed to execute management command 'scanend' (code: 11, error: Rejected)

lukaszbos, 2019-12-07:
Hi,
Changing that line worked in my case
...
devices = scanner.scan(10.0)
...
to
...
devices = scanner.scan(10.0, passive=True)
...

bbostock, 2019-12-27:
I'm getting the same problem on Raspberry Pi with Raspbian, I treed adding "passive=true", but it no longer finds the devices. Any other ideas?

dima72, 2020-09-11:
> Try `sudo hciconfig hci0 down && sudo hciconfig hci0 up`.

worked for me on rpiZeroW
to solve

_bluepy.btle.BTLEManagementError: Failed to execute management command 'scanend' (code: 13, error: Invalid Parameters)_

Moulde, 2021-01-19:
> > Try `sudo hciconfig hci0 down && sudo hciconfig hci0 up`.
> 
> worked for me on rpiZeroW
> to solve
> 
> _bluepy.btle.BTLEManagementError: Failed to execute management command 'scanend' (code: 13, error: Invalid Parameters)_

This seems to have fixed my issue with rejected as well.
`bluepy.btle.BTLEManagementError: Failed to execute management command 'scanend' (code: 11, error: Rejected)`

I detect the error in my code and run the above commands in my python script like this.
`resp = subprocess.call("hciconfig hci0 down && hciconfig hci0 up", shell=True)`

PythonLinWencai, 2021-07-26:
大神们，请给个详细方案

chrismyers81, 2022-06-05:
> Try `sudo hciconfig hci0 down && sudo hciconfig hci0 up`.

This seems to have, at least temporarily, worked around the issue for me.
Trying to read an airthings waveplus via a raspberry pi 3b+ every 15 minutes, posting the results to emoncms. This will sometimes work for weeks before failing, and sometimes fail after a few hours/days.

mklsbali, 2023-05-10:
For me

sudo apt update
sudo apt upgrade

solved the issue.