IanHarvey/bluepy #152: Read and/or Notify

danielet, 2016-09-21:
Hi all, 
I don't know if it is a configuration problem, bluepy or something else.

I can read/write/notify the BLE characteristics between the Arduino board (peripheral role) - BLE device (Feather made by Adafruit)- and my custom iOS APP (central role)

So I decided to use Rasperry Pi 3 to become the GW for the Arduino board.
The Raspeberry runs Jessie kernel 4.4.21, bluez 0.39 and last version of bluepy cloned by git.

I am able to: 
    - connect
    - scan the services
    - scan the characteristics 

Therefore my last step was to read the data. Here I had the surprise that when I call the read function from the characteristic the string/message is empty.

So I checked through bluetoothctl and I used the service-attribute to try to read the data from a specific charateristic. I was surpirese that I read the data.

Here I reported a part of the sourcecode:

`
p = Peripheral(d)
service=p.getServiceByUUID(serviceUUID)
print(service)

for char in chars:
      print(char)
   char=service.getCharacteristics(tempUUID)
   if(char !=[]):
      print("FOUND TEMPERATURE CHARACTERISTIC")
    print(char[0].getHandle())
    print(char[0].propertiesToString())
    while(True):
    if(char[0].supportsRead()):
         print(char[0].read())`

Do you think the problem is my code?
THanks

danielet, 2016-09-22:
Ok, my fault! 
I figured out the problem and for sure it does not regard the library

closed 2016-09-22 by danielet

kmvinayaka, 2017-12-28:
please share code if resolved