IanHarvey/bluepy #159: Work with notifications

jwillyfg, 2016-10-25:
Hi!

I'm not sure how notifications works on this API, I would like to get async notifications, but on each example i've read, i always find a call to WaitforNotification() method. 

I would not like to block my code until a notification is recived.... Does anyone know how i can get async notifications without using WaitforNotifiacion() method.

Thanks so much!

EinSoldiatGott, 2016-12-10:
I've the same question. It's supposed to be async in the declaration. But We've to wait, so it turns to be sync. I don't understand

PrzemoF, 2017-02-20:
Create a separate thread to handle the BLE device. That separate thread can wait for notifications and feed the main thread with received data.

That's something like your main program:
https://github.com/PrzemoF/Open-Cycling-Computer/blob/master/code/src/test_ble.py
And that's the BLE part that waits for notifications:
https://github.com/PrzemoF/Open-Cycling-Computer/blob/master/code/src/ble.py

closed 2017-09-01 by IanHarvey

Snevzor, 2018-01-08:
Hello @PrzemoF. Those examples seem to be inaccessible. Do you still have them and would you like to share them again please?

PrzemoF, 2018-01-08:
Hi @Snevzor! I can't easily do it right now, but if you check the OpenCyclongComputer repo you'll find the files - look for files with "ble" in name.

Snevzor, 2018-01-08:
Thanks @PrzemoF! I've located the examples [here](https://github.com/PrzemoF/Open-Cycling-Computer/tree/master/code/src). Really nice project and great examples! Unfortunately I was hoping to see some 'parallel' execution of getchar and waitfornotifications where I ([and perhaps Bluepy)](https://github.com/IanHarvey/bluepy/issues/250) seem to be struggling.

PrzemoF, 2018-01-08:
Thanka! OCC uses both ways - notificationa run in a separate thread ("run") and it also uses direct reading of characteristic ("get_device_name")
P.S. I wrote the abovw on my phone - I can't point you dotectly to the code.

Snevzor, 2018-01-10:
And you've never run into race condition issues? I'm struggling with [250](https://github.com/IanHarvey/bluepy/issues/250)