IanHarvey/bluepy #160: [Errno 2] No such file or directory

soulraven, 2016-11-23:
Hello, i have try to connect using your lib to a CC2650 SensorTag but i get this error:
`Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/opt/pyProjectsEnv/weatherStation/local/lib/python2.7/site-packages/bluepy/sensortag.py", line 331, in __init__
    Peripheral.__init__(self,addr)
  File "/opt/pyProjectsEnv/weatherStation/local/lib/python2.7/site-packages/bluepy/btle.py", line 318, in __init__
    self.connect(deviceAddr, addrType, iface)
  File "/opt/pyProjectsEnv/weatherStation/local/lib/python2.7/site-packages/bluepy/btle.py", line 353, in connect
    self._startHelper()
  File "/opt/pyProjectsEnv/weatherStation/local/lib/python2.7/site-packages/bluepy/btle.py", line 210, in _startHelper
    universal_newlines=True)
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
`

any ideea i am using BT adaptor ASUS USB-BT400

any idea about this error?

rytilahti, 2016-12-19:
Install from git if possible (pip install git+https://github.com/IanHarvey/bluepy.git), or compile bluepy-helper manually, maybe that fixes it?

blendmaster, 2017-04-08:
I had the same issue with the the regular `pip install bluepy` version (on ubuntu 16.04 / python 2.7 from official repos). I wonder why installing it from git fixes this issue.

abusquets, 2017-08-15:
I have the same problem, also with ubuntu 16.04 and python 2.7

IanHarvey, 2017-09-01:
This has the same symptoms as #158. If I could trouble you to post any further information you have to that issue, it would help.

Thanks
Ian

closed 2017-09-01 by IanHarvey