IanHarvey/bluepy #163: Fast scanning not possible

ISPES, 2016-12-12:
Hey there, 

is there any way to change the default scan behaviour that its possible to scan all packets of a device with 100ms advertising interval? If i do the following:

Initit Scanner, call start(), call process(0)

i get the devices and i also get repeated discovers of the devices. But anyway i dont get all the advertising packets. Are the any possible settings that i can fix that for me? :)

CraigC1, 2016-12-12:
Hey, i'm really not sure if this will help much (if at all) because i'm new to all this, but from a LOT of trial and error i've got something working that will print advertising data from a scan.

`from bluepy.btle import Scanner, DefaultDelegate

class ScanDelegate(DefaultDelegate):
        def init(self):
                DefaultDelegate.__init__(self)

        def handleDiscovery(self, dev, isNewDev, isNewData):
                if isNewDev:
                        print "Discovered device", dev.addr
                        if (dev.addr == "ed:84:9f:3b:6f:ce"):
                                data = dev.getValueText(0xFF)
                                string = data.decode("hex")
                                print string
                                print data

                elif isNewData:
                        print "Received new data from", dev.addr

while True:
        scanner = Scanner().withDelegate(ScanDelegate())
        devices = scanner.scan()
`
The purpose of this is to set my raspberry pi up to constantly scan for any BLE devices in the area and display who they are, and if it's from a device i want the data from, to print that advertising data out.

It seems you can change the 'dev.getValueText() to a lot of other hex codes relating to this: https://www.bluetooth.com/specifications/assigned-numbers/generic-access-profile

At the moment it will run for a random amount of time (up to an hour or so) until it gets an exception "bluepy.btle.BTLEException: Failed to execute mgmt cmd 'scanend'". But looks like someone has figured out a quick fix (issue#150) to stop that happening. 

As an extra, make sure you're running it as sudo (if it's a pi) and also Ian mentions that you can turn debugging to 'True' in btle.py to get much more info from the packet (think it actually shows you the whole packet in hex).

Hope this helps somewhat.

aeyuxi, 2018-01-25:
To get the data, the code is 0x16. Take a look [here](https://www.bluetooth.com/specifications/assigned-numbers/generic-access-profile). Also, that data is in hex. You will have to use binascii.unhexlify(hex data) to  be readable. Also, the UUID is concatenated with the data. You will have to remove it.