IanHarvey/bluepy #164: Notifications not working with VBttn

Catpants, 2017-01-10:
Hi, I am trying to connect a [vbttn](http://vsnmobil.com/products/v-alrt/) device so I can get notifications (specifically button presses). I have followed the notification tutorial but have not been able to receive notifications from the vbttn device. However It does connect to the device. Here is the code:

import btle

class MyDelegate(btle.DefaultDelegate):
    def __init__(self, params):
        btle.DefaultDelegate.__init__(self)
        # ... initialise here

    def handleNotification(self, cHandle, data):
        print("Handling notification")
        # ... perhaps check cHandle
        # ... process 'data'

# Initialisation  -------
print("Connecting to device")
p = btle.Peripheral("D0:39:72:C5:76:CC")    
params = "data"
d = MyDelegate(params)
p.setDelegate(d)

# Setup to turn notifications on, e.g.
print("Sending verification token")
ch = p.getCharacteristics(uuid="FFFFFFF5-00F7-4000-B000-000000000000")[0]
ch.write(bytearray([0x80, 0xBE, 0xF5, 0xAC, 0xFF]))

print("Enabling press button")
ch = p.getCharacteristics(uuid="FFFFFFF2-00F7-4000-B000-000000000000")[0]
ch.write(bytearray([0x01]))

print("Subscribing to press button")
p.writeCharacteristic(0x2902, bytearray([0x01]))
                                                               

# Main loop --------

while True:
    if p.waitForNotifications(10.0):
        print("Loop notification")
        # handleNotification() was called
        continue

    print("Waiting...")
    # Perhaps do something else here

bridger, 2017-01-31:
I'm having the same issue. Did you find a workaround?

I also found that when I scan I don't get updates as often as I'd expect. My peripheral is broadcasting every 100ms, but I only get scan updates every few seconds when not connected. I wonder if it is related?

bridger, 2017-03-14:
I figured out why I was having an issue. It turns out that notifications need to be turned on for different characteristics. For characteristics that support notifications then writing 0100 to the right handle turns on the notifications. I'm not sure how to find the right notification handle, though. In two cases I saw it was the read handle + 1. I don't know if that is always true.

My code, which is reading the scratch characteristic from a [LightBlue Bean](https://punchthrough.com/bean):

	  def listenBluetooth():
	      # Connection loop
	      while True:
	          hammer = None
	          try:
	              print "Connecting to bluetooth"
	              hammer = btle.Peripheral(DEVICE_ID)
	          except Exception as exc:
	              print "Error connecting to bluetooth hammer", exc
	              print "Bluetooth thread sleeping 10 seconds"
	              time.sleep(10)
	              continue
        
	          # Read data loop
	          try:
	              scratchServices = hammer.getServiceByUUID('a495ff20-c5b1-4b44-b512-1370f02d74de')
	              scratch1 = scratchServices.getCharacteristics('a495ff21-c5b1-4b44-b512-1370f02d74de')[0]
	              # Enable notifications on the scratch characteristic
	              hammer.writeCharacteristic(scratch1.valHandle + 1, struct.pack('<bb', 0x01, 0x00))
            
	              lastMotionValue = 0
            
	              while True:
	                  data = scratch1.read()
	                  if len(data) == 4:
	                      (value, ) = struct.unpack('<i', data)
	                      if value == 1 and lastMotionValue == 0:
	                          activateAnimation("thorlightning")
	                      lastMotionValue = value
	                  hammer.waitForNotifications(30)
                
	          except Exception as exc:
	              print "Error reading bluetooth data", exc
	              print "Bluetooth thread sleeping 5 seconds"
	              time.sleep(5)
	              hammer.disconnect()
	              del hammer
	              continue

	  thread.start_new_thread(listenBluetooth, ())

ArnaudLoos, 2017-10-17:
Here is working code to receive notifications from a v.Bttn

```
from bluepy.btle import *
import time
import struct

class ScanDelegate(DefaultDelegate):
    def __init__(self):
        DefaultDelegate.__init__(self)

    def handleNotification(self, cHandle, data):
        val_data = struct.unpack("B"*len(data),data)
        if val_data[0] == 1:
            print "Button press"
        elif val_data[0] == 0:
            print "Button release"
        elif val_data[0] == 3:
            print "Long button press"
        elif val_data[0] == 4:
            print "fall detected"

myVbttn = None
notifOn = b"\x01\x00"
notifOff = b"\x01\x00"

scanner = Scanner().withDelegate(ScanDelegate())
devices = scanner.scan(5.0)

for dev in devices:
    print "Device %s (%s), RSSI=%d dB" % (dev.addr, dev.addrType, dev.rssi)
    for (adtype, desc, value) in dev.getScanData():
        if (desc == "Complete Local Name") and (value[:6] == "V.ALRT") :
            myVbttn = dev
            print "\nDevice %s is a v.bttn\n" % (myVbttn.addr)

if myVbttn is not None:
    print("Connecting to vBttn...")
    try:
        p = Peripheral(myVbttn.addr,myVbttn.addrType).withDelegate(ScanDelegate())
    except:
        print("Error")
    print "\tConnected to ", myVbttn.addr

    item = p.getCharacteristics()

    for k in item:
        if k.uuid==AssignedNumbers.batteryLevel:
            bat_char = k
    bat_data = bat_char.read()
    bat_data_value = ord(bat_data[0])
    print "\tBattery: ", bat_data_value, "%"

    print("Sending verification token...")
    ch = p.getCharacteristics(uuid="FFFFFFF5-00F7-4000-B000-000000000000")[0]
    ch.write(bytearray([0x80, 0xBE, 0xF5, 0xAC, 0xFF]))
    print("\tVerified")

    print("Enabling button press...")
    ch = p.getCharacteristics(uuid="FFFFFFF2-00F7-4000-B000-000000000000")[0]
    ch.write(bytearray([0x07]))
    print("\tEnabled")

    print("Enabling notifications...")
    char = p.getCharacteristics(uuid="fffffff4-00f7-4000-b000-000000000000")[0]
    ccc_desc = char.getDescriptors(forUUID=0x2902)[0]
    ccc_desc.write(notifOn, withResponse=True)
    print("\tDone")

else:
    print("myVbttn reporting as None")

while True:
    if p.waitForNotifications(1.0):
        continue
    print("Waiting...")
```