IanHarvey/bluepy #167: BTLE.py don't work

raf6410, 2017-01-25:
Hello ,

I have install Bluepy last weekend and after follow all the explaination for the installation of this packages, i don't finf a way to use BTLE.py.

Thisis the result when i execute the script :
_pi@raspberrypi:~/bluepy/bluepy $ sudo python btle.py E9:39:68:32:10:28
Connecting to: E9:39:68:32:10:28, address type: public
Running  /home/pi/bluepy/bluepy/bluepy-helper
Sent:  conn E9:39:68:32:10:28 public

Got: "rsp=$stat state=$tryconn dst='E9:39:68:32:10:28 mtu=h0 sec='low\n"
Got: "rsp=$stat state=$disc mtu=h0 sec='low\n"
Stopping  /home/pi/bluepy/bluepy/bluepy-helper
Traceback (most recent call last):
  File "btle.py", line 721, in <module>
    conn = Peripheral(devAddr, addrType)
  File "btle.py", line 353, in __init__
    self.connect(deviceAddr, addrType, iface)
  File "btle.py", line 402, in connect
    "Failed to connect to peripheral %s, addr type: %s" % (addr, addrType))
__main__.BTLEException: Failed to connect to peripheral E9:39:68:32:10:28, addr type: public_

In the same time i have execute a dump : 
sudo hcidump -t
HCI sniffer - Bluetooth packet analyzer ver 5.23
device: hci0 snap_len: 1500 filter: 0xffffffff
2017-01-25 19:13:46.825728 
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
	type 0x00 (passive)
	interval 60.000ms window 30.000ms
    own address: 0x00 (Public) policy: white list only
2017-01-25 19:13:46.826441 
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2017-01-25 19:13:46.826473 
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
2017-01-25 19:13:46.826917 
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2017-01-25 19:14:26.845262 
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)
2017-01-25 19:14:26.847460 
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
	

So i search un the script i stop in : _def _waitResp(self, wantType, timeout=None):_
I thinks that I dont find all "helper" descriptor.

Anyone could you help be to solve my problem?

Tks

Raf6410

raf6410, 2017-02-04:
Hello,

I make some test and the stript stop at this rows : 
_rv = self._helper.stdout.readline()_

Below the debug of  BTLE.py

pi@raspberrypi:~/BLE/bluepy/bluepy $ sudo python btle.py eb:57:31:f3:3c:30
Connecting to: eb:57:31:f3:3c:30, address type: public
Running  /home/pi/BLE/bluepy/bluepy/bluepy-helper
Sent:  conn eb:57:31:f3:3c:30 public

      _waitResp before the error row
                     error row :      rv = self._helper.stdout.readline()
      _waitResp aftee the error row
Got: "rsp=$stat state=$tryconn dst='eb:57:31:f3:3c:30 mtu=h0 sec='low\n"
      _waitResp before the error row
error row : rv = self._helper.stdout.readline()

Got: "rsp=$stat state=$disc mtu=h0 sec='low\n"
Stopping  /home/pi/BLE/bluepy/bluepy/bluepy-helper
Traceback (most recent call last):
  File "btle.py", line 727, in <module>
    conn = Peripheral(devAddr, addrType)
  File "btle.py", line 356, in __init__
    self.connect(deviceAddr, addrType, iface)
  File "btle.py", line 409, in connect
    "Failed to connect to peripheral %s, addr type: %s" % (addr, addrType))
__main__.BTLEException: Failed to connect to peripheral eb:57:31:f3:3c:30, addr type: public

Anyone could you help be to solve my problem?

Tks

Raf6410

IanHarvey, 2017-06-19:
Devices with the most significant address bits set are 'random' addresses, so 'eb:xx:xx' is a random address (for reference: Bluetooth 4.0 spec, Volume 3, Part C, Section 10.8 !). 

You should be able to connect with `python btle.py  eb:57:31:f3:3c:30 random`  (although of course this MAC address may now have changed...)

Did this help?

raf6410, 2017-06-19:
thanks, it better like this.
I miss to add " random" after address devices.

IanHarvey, 2017-06-19:
Thanks for letting me know, I'll close this issue.

Cheers
Ian

closed 2017-06-19 by IanHarvey